IRMP-org/IRMP #5: Wrong / misleading input pin definition for STM32 in all examples

BugerDread, 2020-04-05:
Hi,
I was trying to make IRMP version 1.2.2 working with Arduino 1.8.12 on Bluepill (STM32 cores v1.8.0), but none of your examples worked, because there is wrong input pin definition for STM32 in all of them:

```
#elif defined(STM32F1xx) || defined(__STM32F1__)
// BluePill in 2 flavors
// STM32F1xx is for "Generic STM32F1 series" from STM32 Boards from STM32 cores of Arduino Board manager
// __STM32F1__is for "Generic STM32F103C series" from STM32F1 Boards (STM32duino.com) of manual installed hardware folder
#define IRMP_INPUT_PIN 4 // PA4
#define BLINK_13_LED_IS_ACTIVE_LOW // The LED on the BluePill is active LOW
```
The problem is that PA4 is not pin 4 but 24 (or simply PA4).

You need to change this line:
`#define IRMP_INPUT_PIN 4 // PA4`
to
`#define IRMP_INPUT_PIN PA4 // PA4`
or (but whyyy?)
`#define IRMP_INPUT_PIN 24 // PA4`

Then all examples work as expected. 

This most probably apply to IRMP v2.0.0 also, as there is the same definition for example here: https://github.com/ukw100/IRMP/blob/737c49f4e8152a744f137bb6a805f1240d58b5ee/examples/SimpleReceiver/SimpleReceiver.ino#L62

Please fix the examples, I wasted about half an hour trying to find out why it does not work.

ArminJo, 2020-04-05:
Hi,
thank you for finding this bug, it is fixed in the repo now.
With STM32duino.com the plain 4 does work, but with STM32 cores it is definitely wrong!

And half an hour is a good rate, for me such errors took me often one or more hours.

ArminJo, 2020-04-06:
Release with bug fix created.

closed 2020-04-06 by ArminJo