IRMP-org/IRMP #9: AllProtocols compilation errors and warnings with serial LCD

gitterman, 2020-07-12:
Yesterday I installed this promising project with the library manager in the arduino IDE 1.18.12.
I attached an IR detector and a I2C display to an arduino nano as shown in the pictures.
I tested the LCD with a hello world sketch and it worked at the default address 0x27.
I tried the simpleReceiver and this also worked well but I want to build a standalone IR remote detector.
So I compiled the AllProtocols example with the USE_SERIAL_LCD define and get the errors below.
```
In file included from D:\Peter\Development\Arduino\libraries\IRMP\src/irmp.c.h:26:0,
from D:\Peter\Development\Arduino\AllProtocols\AllProtocols.ino:54:
D:\Peter\Development\Arduino\libraries\IRMP\src/irmp.h:131:4: warning: #warning MERLIN protocol disabled, IRMP_32_BIT=1 needed [-Wcpp]
# warning MERLIN protocol disabled, IRMP_32_BIT=1 needed
^~~~~~~
D:\Peter\Development\Arduino\libraries\IRMP\src/irmp.h:243:4: warning: #warning F_INTERRUPTS too low, LEGO protocol disabled (should be at least 20000) [-Wcpp]
# warning F_INTERRUPTS too low, LEGO protocol disabled (should be at least 20000)
^~~~~~~
D:\Peter\Development\Arduino\libraries\IRMP\src/irmp.h:279:4: warning: #warning F_INTERRUPTS too low, RCMM protocol disabled (should be at least 20000) [-Wcpp]
# warning F_INTERRUPTS too low, RCMM protocol disabled (should be at least 20000)
^~~~~~~
AllProtocols:63:32: error: 'LCD_COLUMNS' was not declared in this scope
LiquidCrystal_I2C myLCD( 0x27, LCD_COLUMNS, LCD_ROWS ) ; // set the LCD address to 0x27 for a 20 chars and 2 line display
^~~~~~~~~~~
D:\Peter\Development\Arduino\AllProtocols\AllProtocols.ino:63:32: note: suggested alternative: 'LCD_2LINE'
LiquidCrystal_I2C myLCD( 0x27, LCD_COLUMNS, LCD_ROWS ) ; // set the LCD address to 0x27 for a 20 chars and 2 line display
^~~~~~~~~~~
LCD_2LINE
AllProtocols:63:45: error: 'LCD_ROWS' was not declared in this scope
LiquidCrystal_I2C myLCD( 0x27, LCD_COLUMNS, LCD_ROWS ) ; // set the LCD address to 0x27 for a 20 chars and 2 line display
^~~~~~~~
D:\Peter\Development\Arduino\AllProtocols\AllProtocols.ino: In function 'void setup()':
AllProtocols:106:11: error: 'class LiquidCrystal_I2C' has no member named 'init'
myLCD.init() ;
![i2c](https://user-images.githubusercontent.com/10762929/87241092-f1eb9580-c41f-11ea-8e18-b13d43cc50ca.jpg)

```

ArminJo, 2020-07-12:
And what do you think is missing?

gitterman, 2020-07-12:
Sorry for not being clear enough.
The "AllProtocols" example does not compile so it doesn't work.

ArminJo, 2020-07-12:
And the error messages are there for fun und to confuse the russians?

ArminJo, 2020-07-12:
Hint: look in the PinDefinitionsAndMore.h file.

gitterman, 2020-07-12:
Thanks for the hint,
I already had enabled USE_1602_LCD before but it also gave an error so I did comment it again.
The real error was the last line which says:
T:\TMP\arduino_modified_sketch_501943\AllProtocols.ino: In function 'void setup()':
AllProtocols:115:11: error: 'class LiquidCrystal_I2C' has no member named 'init'
     myLCD.init();
I found that I had an old version of the LiquidCrystal_I2C library which I had installed manually 2 years ago.
After I have removed it and installed the library using the library manager it now works like a charm

closed 2020-07-12 by gitterman

ArminJo, 2020-07-12:
Thanks for the advice 👍 
I added a comment to use the new library version.