IRMP-org/IRMP #12: interrupt mode do not compile with stm32

pat1, 2020-08-12:
## Bug Report

### Arduino Platform
* [ ] Arduino ATmega328* board (UNO, Nano)
* [ ] Arduino ATmega2560 board (Mega)
* [ ] Arduino ATmega32U4 board (Leonardo)
* [ ] Arduino megaAVR board (NanoEvery)
* [ ] Arduino SAM board (Due)
* [ ] Arduino SAMD board (Zero, MKR*)
* [ ] ESP8266 board
* [ ] ESP32 board
* [ ] Apollo3 board
* [ ] Teensy board
* [X] Other - please specify
STM32duino

### IDE
* [ ] Arduino IDE
* [ ] Arduino Pro IDE
* [ ] Sloeber IDE
* [X] PlatformIO IDE
* [ ] Other - please specify

### IR-Protocol
* [ ] Unknown
* [ ] Sony
* [X] NEC
* [ ] Samsung
* [ ] RC5, RC6
* [ ] Kaseikyo
* [ ] Denon
* [ ] JVC
* [ ] Lego
* [ ] Matsushita
* [ ] Mitsubishi
* [ ] Panasonic
* [ ] Sanyo
* [ ] Sharp
* [ ] Telefunken
* [ ] Other - please specify

### Example to reproduce the issue
* [ ] AllProtocols
* [ ] Callback
* [X] Interrupt
* [ ] OneProtocol
* [ ] ReceiveAndSend
* [ ] ReceiveAndSendDynamicPins
* [ ] RFProtocols
* [ ] SimpleReceiver
* [ ] SimpleSender
* [ ] Other

### Pin(s) used for IR-receive, if not default

### Version
* [X] Yes I use the [latest release](https://github.com/ukw100/IRMP/releases)
* [ ] Other - please specify

### Current behavior

```
arm-none-eabi-g++ -o .pio/build/nucleo_l432kc/src/irtranslator.ino.cpp.o -c -std=gnu++14 -fno-threadsafe-statics -fno-rtti -fno-exceptions -fno-use-cxa-atexit -fexceptions -mfpu=fpv4-sp-d16 -mfloat-abi=hard -Os -mcpu=cortex-m4 -mthumb -ffunction-sections -fdata-sections -Wall -nostdlib --param max-inline-insns-single=500 -DPLATFORMIO=40400 -DSTM32L432xx -DSTM32L4xx -DARDUINO=10808 -DARDUINO_ARCH_STM32 -DARDUINO_NUCLEO_L432KC -DBOARD_NAME=\"NUCLEO_L432KC\" -DHAL_UART_MODULE_ENABLED -Isrc -I.pio/libdeps/nucleo_l432kc/IRMP/src -I/home/pat1/git/rmap/arduino/sketchbook/libraries/frtosLog/src -I/home/pat1/git/rmap/arduino/sketchbook/libraries/ArduinoLog -I/home/pat1/git/rmap/arduino/sketchbook/libraries/freertos-cpp/src -I/home/pat1/git/rmap/arduino/sketchbook/libraries/STM32duino_FreeRTOS/src -I/home/pat1/.platformio/packages/framework-arduinoststm32/cores/arduino/avr -I/home/pat1/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32 -I/home/pat1/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/LL -I/home/pat1/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/usb -I/home/pat1/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/OpenAMP -I/home/pat1/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/usb/hid -I/home/pat1/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/usb/cdc -I/home/pat1/.platformio/packages/framework-arduinoststm32/system/Drivers/STM32L4xx_HAL_Driver/Inc -I/home/pat1/.platformio/packages/framework-arduinoststm32/system/Drivers/STM32L4xx_HAL_Driver/Src -I/home/pat1/.platformio/packages/framework-arduinoststm32/system/STM32L4xx -I/home/pat1/.platformio/packages/framework-arduinoststm32/system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I/home/pat1/.platformio/packages/framework-arduinoststm32/system/Middlewares/ST/STM32_USB_Device_Library/Core/Src -I/home/pat1/.platformio/packages/framework-arduinoststm32/system/Middlewares/OpenAMP -I/home/pat1/.platformio/packages/framework-arduinoststm32/system/Middlewares/OpenAMP/open-amp/lib/include -I/home/pat1/.platformio/packages/framework-arduinoststm32/system/Middlewares/OpenAMP/libmetal/lib/include -I/home/pat1/.platformio/packages/framework-arduinoststm32/system/Middlewares/OpenAMP/virtual_driver -I/home/pat1/.platformio/packages/framework-cmsis/CMSIS/Core/Include -I/home/pat1/.platformio/packages/framework-arduinoststm32/system/Drivers/CMSIS/Device/ST/STM32L4xx/Include -I/home/pat1/.platformio/packages/framework-arduinoststm32/system/Drivers/CMSIS/Device/ST/STM32L4xx/Source/Templates/gcc -I/home/pat1/.platformio/packages/framework-cmsis/CMSIS/DSP/Include -I/home/pat1/.platformio/packages/framework-arduinoststm32/cores/arduino -I/home/pat1/.platformio/packages/framework-arduinoststm32/variants/NUCLEO_L432KC -I/home/pat1/.platformio/packages/framework-arduinoststm32/variants/NUCLEO_L432KC src/irtranslator.ino.cpp
In file included from .pio/libdeps/bluepill_f103c8/IRMP/src/irmpArduinoExt.cpp.h:22,
                 from .pio/libdeps/bluepill_f103c8/IRMP/src/irmp.c.h:3359,
                 from /home/pat1/git/rmap/platformio/domotica/freertos_irtranslator/src/irtranslator.ino:36:
.pio/libdeps/bluepill_f103c8/IRMP/src/irmpPinChangeInterrupt.cpp.h: In function 'void initPCIInterrupt()':
.pio/libdeps/bluepill_f103c8/IRMP/src/irmpPinChangeInterrupt.cpp.h:197:5: error: 'EICRA' was not declared in this scope
  197 |     EICRA |= _BV(ISC10);
      |     ^~~~~
In file included from /home/pat1/.platformio/packages/framework-arduinoststm32/cores/arduino/wiring.h:34,
                 from /home/pat1/.platformio/packages/framework-arduinoststm32/cores/arduino/Arduino.h:36,
                 from /tmp/tmp0h4v60xc:1:
.pio/libdeps/bluepill_f103c8/IRMP/src/irmpPinChangeInterrupt.cpp.h:197:18: error: 'ISC10' was not declared in this scope
  197 |     EICRA |= _BV(ISC10);
      |                  ^~~~~
/home/pat1/.platformio/packages/framework-arduinoststm32/cores/arduino/wiring_constants.h:99:27: note: in definition of macro '_BV'
   99 |   #define _BV(bit) (1 << (bit))
      |                           ^~~
In file included from .pio/libdeps/bluepill_f103c8/IRMP/src/irmpArduinoExt.cpp.h:22,
                 from .pio/libdeps/bluepill_f103c8/IRMP/src/irmp.c.h:3359,
                 from /home/pat1/git/rmap/platformio/domotica/freertos_irtranslator/src/irtranslator.ino:36:
.pio/libdeps/bluepill_f103c8/IRMP/src/irmpPinChangeInterrupt.cpp.h:199:5: error: 'EIFR' was not declared in this scope
  199 |     EIFR |= 1 << INTF1;
      |     ^~~~
.pio/libdeps/bluepill_f103c8/IRMP/src/irmpPinChangeInterrupt.cpp.h:199:18: error: 'INTF1' was not declared in this scope
  199 |     EIFR |= 1 << INTF1;
      |                  ^~~~~
.pio/libdeps/bluepill_f103c8/IRMP/src/irmpPinChangeInterrupt.cpp.h:201:5: error: 'EIMSK' was not declared in this scope
  201 |     EIMSK |= 1 << INT1;
      |     ^~~~~
.pio/libdeps/bluepill_f103c8/IRMP/src/irmpPinChangeInterrupt.cpp.h:201:19: error: 'INT1' was not declared in this scope
  201 |     EIMSK |= 1 << INT1;
      |                   ^~~~
*** [.pio/build/bluepill_f103c8/src/irtranslator.ino.cpp.o] Error 1
```

pat1, 2020-08-12:
compile with:
#define IRMP_USE_ARDUINO_ATTACH_INTERRUPT 1

I will try if it works

ArminJo, 2020-08-12:
Why have you defined the symbol __AVR__ for this compilation?

pat1, 2020-08-12:
I cannot find __AVR__ in my compilation command:

arm-none-eabi-g++ -o .pio/build/nucleo_l432kc/src/irtranslator.ino.cpp.o -c -std=gnu++14 -fno-threadsafe-statics -fno-rtti -fno-exceptions -fno-use-cxa-atexit -fexceptions -mfpu=fpv4-sp-d16 -mfloat-abi=hard -Os -mcpu=cortex-m4 -mthumb -ffunction-sections -fdata-sections -Wall -nostdlib --param max-inline-insns-single=500 -DPLATFORMIO=40400 -DSTM32L432xx -DSTM32L4xx -DARDUINO=10808 -DARDUINO_ARCH_STM32 -DARDUINO_NUCLEO_L432KC -DBOARD_NAME=\"NUCLEO_L432KC\" -DHAL_UART_MODULE_ENABLED -Isrc -I.pio/libdeps/nucleo_l432kc/IRMP/src -I/home/pat1/git/rmap/arduino/sketchbook/libraries/frtosLog/src -I/home/pat1/git/rmap/arduino/sketchbook/libraries/ArduinoLog -I/home/pat1/git/rmap/arduino/sketchbook/libraries/freertos-cpp/src -I/home/pat1/git/rmap/arduino/sketchbook/libraries/STM32duino_FreeRTOS/src -I/home/pat1/.platformio/packages/framework-arduinoststm32/cores/arduino/avr -I/home/pat1/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32 -I/home/pat1/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/LL -I/home/pat1/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/usb -I/home/pat1/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/OpenAMP -I/home/pat1/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/usb/hid -I/home/pat1/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/usb/cdc -I/home/pat1/.platformio/packages/framework-arduinoststm32/system/Drivers/STM32L4xx_HAL_Driver/Inc -I/home/pat1/.platformio/packages/framework-arduinoststm32/system/Drivers/STM32L4xx_HAL_Driver/Src -I/home/pat1/.platformio/packages/framework-arduinoststm32/system/STM32L4xx -I/home/pat1/.platformio/packages/framework-arduinoststm32/system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I/home/pat1/.platformio/packages/framework-arduinoststm32/system/Middlewares/ST/STM32_USB_Device_Library/Core/Src -I/home/pat1/.platformio/packages/framework-arduinoststm32/system/Middlewares/OpenAMP -I/home/pat1/.platformio/packages/framework-arduinoststm32/system/Middlewares/OpenAMP/open-amp/lib/include -I/home/pat1/.platformio/packages/framework-arduinoststm32/system/Middlewares/OpenAMP/libmetal/lib/include -I/home/pat1/.platformio/packages/framework-arduinoststm32/system/Middlewares/OpenAMP/virtual_driver -I/home/pat1/.platformio/packages/framework-cmsis/CMSIS/Core/Include -I/home/pat1/.platformio/packages/framework-arduinoststm32/system/Drivers/CMSIS/Device/ST/STM32L4xx/Include -I/home/pat1/.platformio/packages/framework-arduinoststm32/system/Drivers/CMSIS/Device/ST/STM32L4xx/Source/Templates/gcc -I/home/pat1/.platformio/packages/framework-cmsis/CMSIS/DSP/Include -I/home/pat1/.platformio/packages/framework-arduinoststm32/cores/arduino -I/home/pat1/.platformio/packages/framework-arduinoststm32/variants/NUCLEO_L432KC -I/home/pat1/.platformio/packages/framework-arduinoststm32/variants/NUCLEO_L432KC src/irtranslator.ino.cpp

Where do you have see __AVR__ defined? I am looking for it

ArminJo, 2020-08-12:
Please use the latest version, your line numbers does not correspond to version 3.0.0.
And check for `#define __AVR__ ` in all your files.

pat1, 2020-08-12:
sorry, I am wrong: 
IRMP @ 2.2.1 has been successfully installed!
platformio do not install the last version

manually update to last git problem was solved!

closed 2020-08-12 by pat1