IRMP-org/IRMP #14: Library not working on STM32L Series

ninadshah18, 2020-08-27:
## Bug Report

### Arduino Platform
* [ ] STM32 board

### IDE
* [ ] Arduino IDE

### IR-Protocol
* [ ] Samsung

### Example to reproduce the issue
* [ ] SimpleSender

### Pin(s) used for IR-receive, if not default

### Version
* [ ] Yes I use the [latest release](https://github.com/ukw100/IRMP/releases) and verified this!

### Current behavior

/*
 *  SimpleSender.cpp
 *
 *  Sends Samsung protocol frames.
 *  Is able to send IR protocol data of 15 main protocols.
 *
 *      Sony SIRCS
 *      NEC + APPLE
 *      Samsung + Samsg32
 *      Kaseikyo
 *
 *      Plus 11 other main protocols
 *      JVC, NEC16, NEC42, Matsushita, DENON, Sharp, RC5, RC6 & RC6A, IR60 (SDA2008) Grundig, Siemens Gigaset, Nokia
 *
 *  To disable one of them or to enable other protocols, specify this before the "#include <irmp.c.h>" line.
 *  If you get warnings of redefining symbols, just ignore them or undefine them first (see Interrupt example).
 *  The exact names can be found in the library file irmpSelectAllProtocols.h (see Callback example).
 *
 *  Copyright (C) 2019-2020  Armin Joachimsmeyer
 *  armin.joachimsmeyer@gmail.com
 *
 *  This file is part of IRMP https://github.com/ukw100/IRMP.
 *
 *  IRMP is free software: you can redistribute it and/or modify
 *  it under the terms of the GNU General Public License as published by
 *  the Free Software Foundation, either version 3 of the License, or
 *  (at your option) any later version.
 *
 *  This program is distributed in the hope that it will be useful,
 *  but WITHOUT ANY WARRANTY; without even the implied warranty of
 *  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 *  GNU General Public License for more details.
 *
 *  You should have received a copy of the GNU General Public License
 *  along with this program.  If not, see <http://www.gnu.org/licenses/gpl.html>.
 *
 */

#include <Arduino.h>

/*
 * Set library modifiers first to set output pin etc.
 */
#include "PinDefinitionsAndMore.h"
//#define IR_OUTPUT_IS_ACTIVE_LOW

#include <irsndSelectMain15Protocols.h>
/*
 * After setting the modifiers we can include the code and compile it.
 */
#include <irsnd.c.h>

//#define STM32F1xx
//#define STM32L4xx
#define IRSND_OUTPUT_PIN 3

IRMP_DATA irsnd_data;

void setup()
{
    Serial.begin(115200);
#if defined(__AVR_ATmega32U4__)
    while (!Serial); //delay for Leonardo, but this loops forever for Maple Serial
#endif
#if defined(SERIAL_USB) || defined(SERIAL_PORT_USBVIRTUAL)
    delay(2000); // To be able to connect Serial monitor after reset and before first printout
#endif
#if defined(__ESP8266__)
    Serial.println(); // to separate it from the internal boot output
#endif

    // Just to know which program is running on my Arduino
    Serial.println(F("START " __FILE__ " from " __DATE__ "\r\nUsing library version " VERSION_IRMP));

    irsnd_init();
    irmp_irsnd_LEDFeedback(true); // Enable send signal feedback at LED_BUILTIN

#if defined(STM32F1xx)
    Serial.println(F("Ready to send IR signals at pin PA5")); // the internal pin numbers are crazy for the STM32 Boards library
#else
    Serial.println(F("Ready to send IR signals at pin " STR(IRSND_OUTPUT_PIN)));
#endif
    irsnd_data.protocol = IRMP_SAMSUNG32_PROTOCOL;
    irsnd_data.address = 0x0707;
    irsnd_data.command = 0xFB04;
    irsnd_data.flags = 0; // repeat frame 0 time

    Serial.print(F("Send 0x"));
    Serial.println(irsnd_data.command, HEX);
    irsnd_send_data(&irsnd_data, false);

}

void loop()
{
    delay(5000);
    uint8_t tNextCommand = irsnd_data.command;
    tNextCommand++;
    // For my Samsung the high byte is the negative of the low byte
    irsnd_data.command = ((~tNextCommand) << 8) | tNextCommand;
 
// #if defined(STM32F1xx)
  //  Serial.println(F("Ready to send IR signals at pin PA5")); // the internal pin numbers are crazy for the STM32 Boards library
//#else
    Serial.println(F("Ready to send IR signals at pin " STR(IRSND_OUTPUT_PIN)));
//#endif

    Serial.print(F("Send 0x"));
    Serial.println(irsnd_data.command, HEX);
    irsnd_send_data(&irsnd_data, false); // This stores timer state and restores it after sending
}

I am expecting IR LED to blink when seen from the camera but it was not blinking.

### Expected behavior

I am expecting that the connecter IR LED should blink when IR code is being emitted.
### Error output

### Additional context

ArminJo, 2020-08-27:
Where do you connect the IR LED? For testing it is easier to connect a standard LED. Did you check other pins (especially pin4)?
Where is the `#define IRSND_OUTPUT_PIN 3` used? Where should it be used?
I just updated the repository. Please using the new repo content for testing.

ninadshah18, 2020-08-28:
I have removed #define IRSND_OUTPUT_PIN 3 and used default pin 4 but still, it is not working. BTQ pin4 for STM32L476 is not a PWM pin. I have pulled the latest repository from master as below but still, it is not blinking.

~/Arduino/libraries/IRMP$ git pull
remote: Enumerating objects: 18, done.
remote: Counting objects: 100% (18/18), done.
remote: Compressing objects: 100% (4/4), done.
remote: Total 18 (delta 14), reused 18 (delta 14), pack-reused 0
Unpacking objects: 100% (18/18), done.
From github.com:ukw100/IRMP
   abca5bc..4664059  master     -> origin/master
Updating abca5bc..4664059
Fast-forward
 .github/issue_template.md                                  |  1 +
 examples/AllProtocols/PinDefinitionsAndMore.h              | 10 +++++++++-
 examples/Callback/PinDefinitionsAndMore.h                  | 10 +++++++++-
 examples/Interrupt/PinDefinitionsAndMore.h                 | 10 +++++++++-
 examples/MinimalReceiver/PinDefinitionsAndMore.h           | 10 +++++++++-
 examples/OneProtocol/PinDefinitionsAndMore.h               | 10 +++++++++-
 examples/RFProtocols/PinDefinitionsAndMore.h               | 10 +++++++++-
 examples/ReceiveAndSend/PinDefinitionsAndMore.h            | 10 +++++++++-
 examples/ReceiveAndSendDynamicPins/PinDefinitionsAndMore.h | 10 +++++++++-
 examples/SimpleReceiver/PinDefinitionsAndMore.h            | 10 +++++++++-
 examples/SimpleSender/PinDefinitionsAndMore.h              | 10 +++++++++-
 src/IRTimer.cpp.h                                          |  3 +++
 12 files changed, 94 insertions(+), 10 deletions(-)

ArminJo, 2020-08-28:
The pins etc. must be defined BEFORE the line #include <irmp.c.h>

ninadshah18, 2020-08-28:
I have already defined it before #include <irsnd.c.h> I feel that the library is not supporting STM32L series, it supports only STM32F1 series. Please, verify.

ArminJo, 2020-08-28:
How and where do you specipy, that you have STM32L Series?

ninadshah18, 2020-08-29:
I haven't specify it anywhere as there is no specific code for STM32L series. So, it takes default configuration and code path. My thinking is STM32L may require some extra processing or configuration which is not supported in current library of IRMP.

ArminJo, 2020-08-29:
OK I added some definitions. Please try the latest version from the repo.

ninadshah18, 2020-09-01:
I have tried with both default pin (pin 4) and custom defined pin (pin 3), but still it didn't work.

ArminJo, 2020-09-07:
I updated the repo.
Please activate #define IRMP_MEASURE_TIMING in PinDefinitionsAndMore.h and connect a LED to this pin. It must flicker.

ninadshahps, 2020-09-09:
I tried #define IRMP_MEASURE_TIMING 3 but still no flickering.

ArminJo, 2020-09-14:
Have you checked the other examples? Just to be sure time is working.

elexal, 2020-09-24:
hi
this library for stm32f1 problem
my code : 
`#include <Arduino.h>
#include "PinDefinitionsAndMore.h"
//#define IRMP_PROTOCOL_NAMES 1
#include <irmpSelectMain15Protocols.h>
#include <irsndSelectMain15Protocols.h>
//#include <irmp.c.h>
#include <irsnd.c.h>
IRMP_DATA irmp_data;
IRMP_DATA irsnd_data;

void setup()
{
  Serial1.begin(115200);
//  irmp_init();
  irsnd_init();
  irmp_irsnd_LEDFeedback(true); // Enable send signal feedback at LED_BUILTIN

  irsnd_data.protocol = IRMP_SAMSUNG32_PROTOCOL;
  irsnd_data.address = 0x0707;
  irsnd_data.command = 0xFB04;
  irsnd_data.flags = 0; // repeat frame 0 time
  irsnd_send_data(&irsnd_data, false);
}

void loop()
{

  delay(500);

    uint8_t tNextCommand = irsnd_data.command;
    tNextCommand++;
    // For my Samsung the high byte is the negative of the low byte
    irsnd_data.command = ((~tNextCommand) << 8) | tNextCommand;
    Serial.print(F("Send 0x"));
    Serial.println(irsnd_data.command, HEX);
    irsnd_send_data(&irsnd_data, false); // This stores timer state and restores it after sending

}`

when #include <irmp.c.h>is activated, there is a problem to send ir & when this line is active it cant send ir code
but receiver both mode is ok !
note: but when they are activated alone in each project they work all right ! but both
#include <irmp.c.h>
#include <irsnd.c.h>
 together not ok !

ArminJo, 2020-09-25:
Thanks for reporting the error.
You forgot to define `USE_ONE_TIMER_FOR_IRMP_AND_IRSND` and the automatic warning failed.
I fixed the warning now in the new release.

commit 20ffe7b9 (2020-09-25): version 3.2.3 Fixed warning for missing `USE_ONE_TIMER_FOR_IRMP_AND_IRSND` macro.
Closes #14
closed 2020-09-25 by ArminJo