IRMP-org/IRMP #15: crash

Riffer, 2020-08-28:
## Bug Report

### Arduino Platform
* [X] ESP8266 board

### IDE
* [X] Arduino IDE
* [X] PlatformIO IDE (VS Code)
(checked with both)

### IR-Protocol
* [X] NEC (probably unrelated)

### Example to reproduce the issue
* [X] SimpleReceiver

### Pin(s) used for IR-receive, if not default
D5 (14)

### Version
* [X] Yes I use the [latest release](https://github.com/ukw100/IRMP/releases) and verified this!
* [X] all above 2.2.1

### Current behavior
Setup with V1838B on ESP8266 W1 Mini (1M) connected to D5 (pin 14) with 3.3 Volt the device crashes reproducable with versions above 2.2.1. With 2.1.1 decoding works, no crash at all.

### Expected behavior
Should work as with 2.2.1 in any newer version.

### Error output
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (3):
epc1=0x4000e041 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4023800d depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3fffec50 end: 3fffffb0 offset: 0190
3fffede0:  4021b278 3fffee80 00000000 40100208  
3fffedf0:  4021b2d9 3fffee80 3fffee80 00000000  
3fffee00:  3fffc200 40100f28 00000020 00000001  
3fffee10:  40100f28 00000020 40100e7c 00000022  
3fffee20:  3fffc200 40100f28 00000020 40100e7c  
3fffee30:  3fffc200 40100f28 00000020 40100f70  
3fffee40:  4022454e 00000020 3fffc230 4000050c  
3fffee50:  40101b50 00000030 00000010 ffffffff  
3fffee60:  40201dac 3fffefa0 00000001 00000000  
3fffee70:  3fffdab0 00000000 3fffd9d0 3ffee3b0  
3fffee80:  00000000 00000000 00000001 3ffe84e8  
3fffee90:  00000000 3fffdad0 3ffee3b0 00000030  
3fffeea0:  00000000 3fffdad0 3ffee3b0 00000030  
3fffeeb0:  00000000 00000000 4bc6a7f0 4020a215  
3fffeec0:  00139ba9 000010f5 00000020 4020a254  
3fffeed0:  00000000 00000000 0000002e 4020af11  
3fffeee0:  4020af6c 000010f5 3ffee4bc 4020af9f  
3fffeef0:  4023a230 0014ae7c 3ffecd80 4020a279  
3fffef00:  00423e19 3ffe8800 3ffe87e4 4020b114  
3fffef10:  40102365 0014ae7c 3ffe8800 60000600  
3fffef20:  40224574 3ffe87e4 3ffe8800 00423e19  
3fffef30:  4022459a 3fffdab0 00000000 3ffe84e8  
3fffef40:  00000000 3fffdad0 3ffee3b0 40201da4  
3fffef50:  40000f49 40000f49 3fffdab0 40000f49  
3fffef60:  40000e19 00000005 00042880 00000000  
3fffef70:  00000000 aa55aa55 000000a1 40101d55  
3fffef80:  40101d5b 00042880 00000000 00000000  
3fffef90:  4010000d 00000000 00000000 00000000  
3fffefa0:  00000000 3fffef3c 00000000 3fffff68  
3fffefb0:  3fffffc0 00000000 00000000 feefeffe  
3fffefc0:  feefeffe feefeffe feefeffe feefeffe  
[... 289 lines cut ...]
0x40100f70: timer1_isr_handler(void*) at C:\Users\kposa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_timer.cpp line 43
0x40100e92: ets_intr_lock() at C:\Users\kposa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 155
0x40100ef0: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\kposa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 177
0x40100ef0: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\kposa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 177
0x40100ef0: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\kposa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 177
0x40100ef0: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\kposa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 177
0x40100208: irmp_ISR() at C:\Users\kposa\Documents\Arduino\libraries\IRMP\src/irmp.c.h line 3211
0x40100208: irmp_ISR() at C:\Users\kposa\Documents\Arduino\libraries\IRMP\src/irmp.c.h line 3211
0x40100f28: timer1_isr_handler(void*) at C:\Users\kposa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_timer.cpp line 36
0x40100208: irmp_ISR() at C:\Users\kposa\Documents\Arduino\libraries\IRMP\src/irmp.c.h line 3211
0x40100f28: timer1_isr_handler(void*) at C:\Users\kposa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_timer.cpp line 36
0x40203399: run_scheduled_functions() at C:\Users\kposa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\Schedule.cpp line 125
0x40100f28: timer1_isr_handler(void*) at C:\Users\kposa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_timer.cpp line 36
0x40100e7c: irmp_timer_ISR() at C:\Users\kposa\Documents\Arduino\libraries\IRMP\src/IRTimer.cpp.h line 622
0x40100f70: timer1_isr_handler(void*) at C:\Users\kposa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_timer.cpp line 43
0x40203374: run_scheduled_functions() at C:\Users\kposa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\Schedule.cpp line 119
0x40201f19: __loop_end() at C:\Users\kposa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 184
0x40202ca5: uart_write(uart_t*, char const*, size_t) at C:\Users\kposa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\uart.cpp line 509
0x40201540: HardwareSerial::write(unsigned char const*, unsigned int) at C:\Users\kposa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/HardwareSerial.h line 164
0x4020154c: HardwareSerial::write(unsigned char const*, unsigned int) at C:\Users\kposa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/HardwareSerial.h line 165
0x40201749: Print::write(char const*) at C:\Users\kposa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/Print.h line 62
0x40201810: Print::println() at C:\Users\kposa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\Print.cpp line 186
0x40100ef0: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\kposa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 177
0x40100f11: esp_schedule() at C:\Users\kposa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 125
0x40201f55: loop_wrapper() at C:\Users\kposa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 199

Riffer, 2020-08-30:
Crash with 3.1.2 and 3.2.0 still there:
----

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (3):
epc1=0x4000e041 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4023800d depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3fffec60 end: 3fffffb0 offset: 0190
3fffedf0:  401025a2 000007a0 00003e80 40100208  
3fffee00:  00000007 00004650 00000022 00000000  
3fffee10:  40100f34 00000020 40100e88 00000022  
3fffee20:  3fffc200 40100f34 40100f7c 00000022  
3fffee30:  3fffc200 40100f34 00000020 40100e88  
3fffee40:  00000030 00000010 ffffffff 40100f7c  
3fffee50:  3fffefa0 40201f28 3fffc230 4000050c  
3fffee60:  40000f83 00000030 00000010 ffffffff  
3fffee70:  40000f3d 00000023 00000000 3fffd9d0  
3fffee80:  00000000 00000000 00000000 fffffffe  
3fffee90:  ffffffff 3fffc6fc 00000001 3fffdab0  
3fffeea0:  00000000 3fffdad0 3ffee3b0 00000030  
3fffeeb0:  00000000 00000000 4bc6a7f0 00000000  
3fffeec0:  401025f5 00333779 401010d4 000029c1  
3fffeed0:  00000000 3ffe9104 00000033 4020af11  
3fffeee0:  003047ac 0032bd67 402341e7 3ffe9104  
3fffeef0:  40102371 0032f3b7 60000600 60000600  
3fffef00:  00a31ba5 3ffe8800 3ffe87e4 4020b0d8  
3fffef10:  40102371 0032f3b7 3ffe8800 60000600  
3fffef20:  40224574 3ffe87e4 3ffe8800 00a31ba5  
3fffef30:  4022459a 3fffdab0 00000000 3ffe84e8  
3fffef40:  00000000 3fffdad0 3ffee3b0 40201da0  
3fffef50:  40000f49 40000f49 3fffdab0 40000f49  
3fffef60:  40000e19 00000005 0004288c 00000000  
3fffef70:  00000000 aa55aa55 000000a1 40101d61  
3fffef80:  40101d67 0004288c 00000000 00000000  
3fffef90:  4010000d 00000000 00000000 00000000  
3fffefa0:  00000000 3fffef3c 00000000 3fffff68  
3fffefb0:  3fffffc0 00000000 00000000 feefeffe  
3fffefc0:  feefeffe feefeffe feefeffe feefeffe  
3fffefd0:  feefeffe feefeffe feefeffe feefeffe  
3fffefe0:  feefeffe feefeffe feefeffe feefeffe  
3fffeff0:  feefeffe feefeffe feefeffe feefeffe  
3ffff000:  feefeffe feefeffe feefeffe feefeffe  
3ffff010:  feefeffe feefeffe feefeffe feefeffe  
3ffff020:  feefeffe feefeffe feefeffe feefeffe  
3ffff030:  feefeffe feefeffe feefeffe feefeffe  
3ffff040:  feefeffe feefeffe feefeffe feefeffe  
3ffff050:  feefeffe feefeffe feefeffe feefeffe  
3ffff060:  feefeffe feefeffe feefeffe feefeffe  
3ffff070:  feefeffe feefeffe feefeffe feefeffe  
3ffff080:  feefeffe feefeffe feefeffe feefeffe  
3ffff090:  feefeffe feefeffe feefeffe feefeffe  
3ffff0a0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0b0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0c0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0d0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0e0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0f0:  feefeffe feefeffe feefeffe feefeffe  
3ffff100:  feefeffe feefeffe feefeffe feefeffe  
3ffff110:  feefeffe feefeffe feefeffe feefeffe  
3ffff120:  feefeffe feefeffe feefeffe feefeffe  
3ffff130:  feefeffe feefeffe feefeffe feefeffe  
3ffff140:  feefeffe feefeffe feefeffe feefeffe  
3ffff150:  feefeffe feefeffe feefeffe feefeffe  
3ffff160:  feefeffe feefeffe feefeffe feefeffe  
3ffff170:  feefeffe feefeffe feefeffe feefeffe  
3ffff180:  feefeffe feefeffe feefeffe feefeffe  
[... 253 lines cut ...]
0x40100e88: irmp_timer_ISR() at C:\Users\kposa\Documents\Arduino\libraries\IRMP\src/IRTimer.cpp.h line 625
0x40100f7c: timer1_isr_handler(void*) at C:\Users\kposa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_timer.cpp line 43
0x40100f7c: timer1_isr_handler(void*) at C:\Users\kposa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_timer.cpp line 43
0x40100f7f: timer1_isr_handler(void*) at C:\Users\kposa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_timer.cpp line 43
0x40100efc: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\kposa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 177
0x40100efc: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\kposa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 177
0x40100efc: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\kposa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 177
0x40100208: irmp_ISR() at C:\Users\kposa\Documents\Arduino\libraries\IRMP\src/irmp.c.h line 3219
0x40100208: irmp_ISR() at C:\Users\kposa\Documents\Arduino\libraries\IRMP\src/irmp.c.h line 3219
0x40100f34: timer1_isr_handler(void*) at C:\Users\kposa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_timer.cpp line 36
0x40100e88: irmp_timer_ISR() at C:\Users\kposa\Documents\Arduino\libraries\IRMP\src/IRTimer.cpp.h line 625
0x40100f7c: timer1_isr_handler(void*) at C:\Users\kposa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_timer.cpp line 43
0x40100f34: timer1_isr_handler(void*) at C:\Users\kposa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_timer.cpp line 36
0x40203435: run_scheduled_recurrent_functions() at C:\Users\kposa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\Schedule.cpp line 157
0x40201f18: __loop_end() at C:\Users\kposa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 185
[... cut ...]
0x4020180c: Print::println() at C:\Users\kposa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\Print.cpp line 186
0x40100efc: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\kposa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 177
0x40100f1d: esp_schedule() at C:\Users\kposa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 125
0x40201f51: loop_wrapper() at C:\Users\kposa\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 199

ArminJo, 2020-08-30:
Strange. Let me look...

ArminJo, 2020-08-30:
Thanks a lot for testing again 👍 
The reason was, that one of the fixes (in digitalWriteFast.h) missed to get into the 3.1.2 release.
So please use the new releases 3.2.0.

Riffer, 2020-08-30:
Many thanks to you for great support on this and your great library.

Any clue wow can I check PlatformIO uses the new 3.2.0?

ArminJo, 2020-08-30:
Check the first output of the examples.
It looks like:
```
START SimpleReceiver.ino from Aug 30 2020
Using library version 3.2.0
```

Riffer, 2020-08-30:
No more crashing with 3.2.0! Thank you very much.

I struggle with using IN and OUT at the same time. Using any from them separately (using #ifdef blocks for compile) they work while having IN active, OUT does no longer work, but this might be a new issue. To circumvent I tried to use interrupt for IN but that failed with totally wrong decoding, maybe another issue, but I will test examples in Arduino 'Studio' just to be sure and open new issues if any.

This one should be declared as solved. Thanks again!

ArminJo, 2020-08-30:
Dit you try the ReceiveAndSend Example?

Riffer, 2020-08-30:
Because I struggle with other things, I did not take the chance, but I will. Thanks for your support!

commit 17751c8f (2020-08-28): Fixed interrupt mode bug introduced by merging upstream code for version 3.1.0 - closes #16. Fixed ESP8266 wrong memcpy_p definition - closes #15
commit 1af9608f (2020-08-28): Fixed interrupt mode bug introduced by merging upstream code for version 3.1.0 - closes #16. Fixed ESP8266 wrong memcpy_p definition - closes #15
closed 2020-08-31 by ArminJo