IRMP-org/IRMP #17: freertos with receive and send IR

pat1, 2020-08-31:
## Question
I try to receive and send IR inside freertos.
I need some suggestions to do a good configuration.
I am working on this example:
https://github.com/r-map/rmap/blob/master/platformio/domotica/freertos_irtranslator/src/freertos_irtranslator.ino

I use NEC protocol and some STM32 board with official stm32duino framework and freertos from a C++ wrapper.
I think that the better configuration is having interrupt for receiving and a dedicated timer to transmit; this for a multitasking environment where I have to spend CPU for other work too. The received messages are queued in a freertos queue consumed by sender task. I now use this configuration:
```
#define IRMP_USE_COMPLETE_CALLBACK       1 // Enable callback functionality
#define IRMP_ENABLE_PIN_CHANGE_INTERRUPT 1 // Enable interrupt functionality
#define USE_ONE_TIMER_FOR_IRMP_AND_IRSND // otherwise we get an error: redefinition of 'void __vector_8()
```
I am not sure about this configuration especially for   USE_ONE_TIMER_FOR_IRMP_AND_IRSND
Do you use a timer in interrupt functionality for IRMP ? If yes I think is better to use two timers for IRMP and IRSND
on STM32 but unsetting USE_ONE_TIMER_FOR_IRMP_AND_IRSND stop the work after the first code received. Witch timer do you use? In documentation I found "For STM32 (BluePill) timer 3 (Servo timer) channel 1 is used as default". But seem to me that for  IRSND you use timer 4. But I need to change the default PINS (I need to use SPI) so witch timer will be used ?

Thanks in advance for any suggestions

Paolo

### Arduino Platform
* [ ] Arduino ATmega328* board (UNO, Nano)
* [ ] Arduino ATmega2560 board (Mega)
* [ ] Arduino ATmega32U4 board (Leonardo)
* [ ] Arduino megaAVR board (NanoEvery)
* [ ] Arduino SAM board (Due)
* [ ] Arduino SAMD board (Zero, MKR*)
* [ ] ESP8266 board
* [ ] ESP32 board
* [X] STM32 board
* [ ] Apollo3 board
* [ ] Teensy board
* [ ] Other - please specify

* bluepill_f103c8
* blackpill_f401cc

### IDE
* [ ] Arduino IDE
* [ ] Arduino Pro IDE
* [ ] Sloeber IDE
* [X] PlatformIO IDE
* [ ] Other - please specify

### IR-Protocol
* [ ] Unknown
* [ ] Sony
* [X] NEC
* [ ] Samsung
* [ ] RC5, RC6
* [ ] Kaseikyo
* [ ] Denon
* [ ] JVC
* [ ] Lego
* [ ] Matsushita
* [ ] Mitsubishi
* [ ] Panasonic
* [ ] Sanyo
* [ ] Sharp
* [ ] Telefunken
* [ ] Other - please specify

### Example to reproduce the issue
* [ ] AllProtocols
* [ ] Callback
* [ ] Interrupt
* [ ] OneProtocol
* [ ] ReceiveAndSend
* [ ] ReceiveAndSendDynamicPins
* [ ] RFProtocols
* [ ] SimpleReceiver
* [ ] SimpleSender
* [ ] Other

### Pin(s) used for IR-receive, if not default

### Version
* [X] Yes I use the [latest release](https://github.com/ukw100/IRMP/releases) and verified this!
* [ ] Other - please specify

ArminJo, 2020-09-07:
Sorry for the late response, but I was busy last week.
I released a new version with STM32 timer bug fixed, this may answer you question, which timer is used.
Since you use IRMP_ENABLE_PIN_CHANGE_INTERRUPT the timer is only used for sending!!!
It should be quite easy to change the timer for STM, but it depends on the rtos / core which timers are available.

pat1, 2020-09-08:
I have found this article https://dzone.com/articles/arm-cortex-m-interrupts-and-freertos-part-3
"SysTick is the default time base. Of course, any other timer interrupt can be used instead. 
....
 And if enabling the FreeRTOS trace facility to measure task execution time, an extra timer might be needed."

closed 2020-09-25 by ArminJo