IRMP-org/IRMP #20: irsnd_send_data() crashes ESP32

sej7278, 2020-09-22:
I can't seem to get sending to work on ESP32's - either my own sketch or the SimpleSender.ino example.

**Software:** Arduino IDE 1.8.13 and arduino-cli 0.13, using esp32 devkit board definition: `fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none`

**Hardware:** TTGO T Display and TTGO T7 mini.

It crashes as soon as you call `irsnd_send_data(&irsnd_data, false);`

```
START /tmp/arduino_modified_sketch_602015/SimpleSender.ino from Sep 22 2020
Using library version 3.2.0
Ready to send IR signals at pin 4
Send 0xFB04
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400ea743  PS      : 0x00060130  A0      : 0x800d0cbc  A1      : 0x3ffb1f10  
A2      : 0x00000000  A3      : 0x3f400fdb  A4      : 0xa5f310fd  A5      : 0x3ffc04d0  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x800d29a1  A9      : 0x3ffb1f00  
A10     : 0x00000002  A11     : 0x3f400fdb  A12     : 0x00000002  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400ea743:0x3ffb1f10 0x400d0cb9:0x3ffb1f30 0x400d149b:0x3ffb1f50 0x400d20d1:0x3ffb1f80 0x400d2ea7:0x3ffb1fb0 0x40088381:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57
```

Receiving works excellently thanks (something that never worked with IRRemote!)

P.S. the ESP32 cores don't seem to define `LED_BUILTIN` so the sketches won't compile even if you set `irmp_irsnd_LEDFeedback(false);` until you define it in your sketch, that's probably more a bug with the cores, but it may be useful to have a default define or ignore if feedback is false in your lib?

ArminJo, 2020-09-23:
Thanks a lot for opening the issue and the stacktrace. 🥇 
The first was a bug, and your P.S. pointed to a design flaw, I corrected.
I built a new release including this bugfix.

closed 2020-09-23 by ArminJo

sej7278, 2020-09-24:
excellent thanks, 3.2.2 works well now - i just made the volume go up on my tv using the code i had read using the library!

next to make a read+write ir "proxy" with a tft display and webserver with a proper pcb and 3d printed case.

ArminJo, 2020-09-24:
Interesting project!
Please send me some pictures if you are finished.
And maybe I can put a link to it in the readme.

sej7278, 2020-09-24:
i certainly will, might be a while yet though ;)

commit b2976a9e (2020-09-23): Fixed bug for sending only on no AVR platforms. Closes #20