IRMP-org/IRMP #21: Detect if IRMP is currently receiving

Roxolasu, 2020-09-26:
## Question / Feature request
### Arduino Platform
* [x] Arduino ATmega328* board (UNO, Nano)

### IDE
* [x] Arduino IDE

### IR-Protocol
* [x] NEC

### Example to reproduce the issue
* [x] Interrupt

### Pin(s) used for IR-receive, if not default

### Version
* [x] Yes I use the [latest release](https://github.com/ukw100/IRMP/releases) and verified this!
* [ ] Other - please specify

### Additional context
I'm trying to use IRMP (using interrupts / callback) in conjunction with code which disables interrupts (NeoPixel). For this, I want to skip the code which disables interrupts if IRMP is currently receiving something.
 
Is there a way to check if IRMP is currently expecting more data? (I tried some flags like irmp_ir_detected, but they didn't seem to have the expected effect)

ArminJo, 2020-09-28:
Use the new function bool irmp_IsBusy()

closed 2020-09-28 by ArminJo

Roxolasu, 2020-09-28:
Thanks!

Unfortunately, this doesn't seem to work: 
Given the interrupt example, with the following loop function
`if(irmp_IsBusy()){
    Serial.println("Busy");
else{
    Serial.println("Not Busy");
}
delay(100);
`
This always prints busy.

ArminJo, 2020-09-28:
Congratulation. 🥇 You tested the case I missed.
I updated the code with a version which also gives reliable results for interrupt mode 😀 .

commit 3df03b20 (2020-09-28): Version 3.3.0 Added function `bool irmp_IsBusy()`. Closes #21
commit b4e43e51 (2020-09-28): Version 3.3.0 Added function `bool irmp_IsBusy()`. Closes #21
commit bbc1a000 (2020-09-28): Version 3.3.0 Added function `bool irmp_IsBusy()`. Closes #21
commit 5afd72b7 (2020-09-28): Version 3.3.0 Added function `bool irmp_IsBusy()`. Closes #21