IRMP-org/IRMP #22: proto_rf_medion missing from irmp_protocol_names

sej7278, 2020-09-29:
`proto_rf_medion` is missing from `irmp_protocol_names` in 3.3.1 (its the only one, i went through them all as part of writing my app!); so `P=59` doesn't get converted to `P=RF_MEDION`, here's the one-liner fix - didn't bother with a PR:

```diff
simon@greivous:~/IRMP/src$ git diff
diff --git a/src/irmp.c.h b/src/irmp.c.h
index d20c214a..e7a46a99 100644
--- a/src/irmp.c.h
+++ b/src/irmp.c.h
@@ -835,7 +835,8 @@ irmp_protocol_names[IRMP_N_PROTOCOLS + 1] PROGMEM =
     proto_onkyo,
 
     proto_rf_gen24,
-    proto_rf_x10
+    proto_rf_x10,
+    proto_rf_medion
 };
 
 #endif
```

ArminJo, 2020-09-29:
Thanks a lot!
I will wait a few days before the new bugfix Release.

closed 2020-09-29 by ArminJo

sej7278, 2020-09-29:
no problem, thanks for the lib!

commit 5bd8bcf9 (2020-09-29): Added missing Medion entry in irmp_protocol_names. Closes #22