IRMP-org/IRMP #23: Nano 33 BLE - IRSND does not work

tomaculum, 2020-09-29:
## Bug Report

### Arduino Platform
* [ ] Arduino ATmega328* board (UNO, Nano)
* [ ] Arduino ATmega2560 board (Mega)
* [ ] Arduino ATmega32U4 board (Leonardo)
* [ ] Arduino megaAVR board (NanoEvery)
* [ ] Arduino SAM board (Due)
* [ ] Arduino SAMD board (Zero, MKR*)
* [ ] ESP8266 board
* [ ] ESP32 board
* [ ] STM32 board with "STM32 Boards (selected from submenu)" core
* [ ] STM32 board with "STM32F1 Boards (STM32duino.com)" core
* [ ] Apollo3 board
* [ ] Teensy board
* [x] Other - please specify

Arduino Nano 33 BLE

### IDE
* [x] Arduino IDE
* [ ] Arduino Pro IDE
* [ ] Sloeber IDE
* [ ] PlatformIO IDE
* [ ] Other - please specify

### IR-Protocol
* [ ] Unknown
* [ ] Sony
* [x] NEC
* [ ] Samsung
* [x] RC5, RC6
* [ ] Kaseikyo
* [ ] Denon
* [ ] JVC
* [ ] Lego
* [ ] Matsushita
* [ ] Mitsubishi
* [ ] Panasonic
* [ ] Sanyo
* [ ] Sharp
* [ ] Telefunken
* [ ] Other - please specify

### Example to reproduce the issue
* [ ] AllProtocols
* [ ] Callback
* [ ] Interrupt
* [ ] OneProtocol
* [ ] ReceiveAndSend
* [ ] ReceiveAndSendDynamicPins
* [ ] RFProtocols
* [ ] SimpleReceiver
* [x] SimpleSender
* [ ] Other

### Pin(s) used for IR-receive, if not default

### Version
* [x] Yes I use the [latest release](https://github.com/ukw100/IRMP/releases) and verified this!
* [ ] Other - please specify

### Current behavior

<details>
  <summary>Code</summary>

``` c++
/*
 *  SimpleSender.cpp
 *
 *  Sends Samsung protocol frames.
 *  Is able to send IR protocol data of 15 main protocols.
 *
 *      Sony SIRCS
 *      NEC + APPLE
 *      Samsung + Samsg32
 *      Kaseikyo
 *
 *      Plus 11 other main protocols
 *      JVC, NEC16, NEC42, Matsushita, DENON, Sharp, RC5, RC6 & RC6A, IR60 (SDA2008) Grundig, Siemens Gigaset, Nokia
 *
 *  To disable one of them or to enable other protocols, specify this before the "#include <irmp.c.h>" line.
 *  If you get warnings of redefining symbols, just ignore them or undefine them first (see Interrupt example).
 *  The exact names can be found in the library file irmpSelectAllProtocols.h (see Callback example).
 *
 *  Copyright (C) 2019-2020  Armin Joachimsmeyer
 *  armin.joachimsmeyer@gmail.com
 *
 *  This file is part of IRMP https://github.com/ukw100/IRMP.
 *
 *  IRMP is free software: you can redistribute it and/or modify
 *  it under the terms of the GNU General Public License as published by
 *  the Free Software Foundation, either version 3 of the License, or
 *  (at your option) any later version.
 *
 *  This program is distributed in the hope that it will be useful,
 *  but WITHOUT ANY WARRANTY; without even the implied warranty of
 *  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 *  GNU General Public License for more details.
 *
 *  You should have received a copy of the GNU General Public License
 *  along with this program.  If not, see <http://www.gnu.org/licenses/gpl.html>.
 *
 */

#include <Arduino.h>

/*
 * Set library modifiers first to set output pin etc.
 */
#include "PinDefinitionsAndMore.h"
//#define IR_OUTPUT_IS_ACTIVE_LOW

#include <irsndSelectMain15Protocols.h>
/*
 * After setting the definitions we can include the code and compile it.
 */
#include <irsnd.c.h>

IRMP_DATA irsnd_data;

void setup()
{
    Serial.begin(115200);
#if defined(__AVR_ATmega32U4__)
    while (!Serial); //delay for Leonardo, but this loops forever for Maple Serial
#endif
#if defined(SERIAL_USB) || defined(SERIAL_PORT_USBVIRTUAL)
    delay(2000); // To be able to connect Serial monitor after reset and before first printout
#endif
#if defined(ESP8266)
    Serial.println(); // to separate it from the internal boot output
#endif

    // Just to know which program is running on my Arduino
    Serial.println(F("START " __FILE__ " from " __DATE__ "\r\nUsing library version " VERSION_IRMP));

    irsnd_init();
    irmp_irsnd_LEDFeedback(true); // Enable send signal feedback at LED_BUILTIN

#if defined(ARDUINO_ARCH_STM32)
    Serial.println(F("Ready to send IR signals at pin " IRSND_OUTPUT_PIN_STRING)); // the internal pin numbers are crazy for the STM32 Boards library
#else
    Serial.println(F("Ready to send IR signals at pin " STR(IRSND_OUTPUT_PIN)));
#endif
    irsnd_data.protocol = IRSND_SUPPORT_NEC_PROTOCOL;
    irsnd_data.address = 0xA002;
    irsnd_data.command = 0x80;
    irsnd_data.flags = 0; // repeat frame 0 time

    Serial.print(F("Send 0x"));
    Serial.println(irsnd_data.command, HEX);
    irsnd_send_data(&irsnd_data, false);

}

void loop()
{
    delay(5000);
    uint8_t tNextCommand = irsnd_data.command;
    //tNextCommand++;
    // For my Samsung the high byte is the negative of the low byte
    irsnd_data.command = tNextCommand;
    //irsnd_data.command = ((~tNextCommand) << 8) | tNextCommand;
    Serial.print(F("Send 0x"));
    Serial.println(irsnd_data.command, HEX);
    irsnd_send_data(&irsnd_data, false); // This stores timer state and restores it after sending
}
```
</details>

<details>
  <summary>IRMP output for power button</summary>

``` 
#POWER
P=NEC  A=0xA002 C=0x80
```
</details>

If I am not mistaken, this example with slight changes should send the "POWER" command every five seconds.

### Expected behavior

Toogle power of my device every five seconds.

### Error output

Everything seems fine. The buildin LED flashed every five seconds and the serial output is fine, too.

```
START /Users/tom/workspaces/Arduino/SimpleSender/SimpleSender.ino from Sep 29 2020
Using library version 3.3.1
Ready to send IR signals at pin 4
Send 0x80
Send 0x80
```

### Additional context

I want to replace my LIRC raspberry pi with my arduino therefore I know my layout works and the components are ok, since it works if I plug them into my raspberry.

Reading my original remote with IRMP did work without any issue. 

I did measure the voltage at the IR led and there are peaks every five seconds. Therefore I don't think I am using use the wrong pins.

Maybe there is an issue with the different voltage, since the raspberry does output 5V and the arduino only 3,3V?

I am not using the exact same components as in the documentation, if this is relevant, I can search for the exact specs.

And I did test this with another device and protocol (RC5) but no luck either.

<details>
  <summary>my current layout</summary>

   [image]

</details>

ArminJo, 2020-09-29:
The main problem is, I do not have this board to test the library with it.
On your side everything looks fine, ig you can receive IR signals with the BLE board.
Maybe I find some ways to enable debugging, but this will take some hours...

tomaculum, 2020-09-29:
Yes, the receiving part did work with my Arduino Nano 33 BLE!
Thank you very much, that would be great! And it's ok if it takes days, weeks or even months - there is no pressure of time (;

ArminJo, 2020-10-11:
I am sorry, but I have no idea. Do you have a [logic analyser](https://de.aliexpress.com/wholesale?catId=0&initiative_id=SB_20201011151657&SearchText=SALEAE+24M+8CH+logic+analyzer) or a dso screenshot of the output signal.

closed 2020-10-21 by ArminJo

tomaculum, 2021-02-13:
Hi, just in case someone stumbles across this issue: with version 3.4.0 IRSND works with my Arduino 33 BLE! 
Maybe I did something wrong in the first place or in the meantime some changes did fix this but I am very happy it works (:

ArminJo, 2021-02-13:
Thanks for feedback 👍