IRMP-org/IRMP #24: Teensy 4.0 - cannot detect IR codes

lpaolini, 2020-10-07:
## Bug Report

### Arduino Platform
* [ ] Arduino ATmega328* board (UNO, Nano)
* [ ] Arduino ATmega2560 board (Mega)
* [ ] Arduino ATmega32U4 board (Leonardo)
* [ ] Arduino megaAVR board (NanoEvery)
* [ ] Arduino SAM board (Due)
* [ ] Arduino SAMD board (Zero, MKR*)
* [ ] ESP8266 board
* [ ] ESP32 board
* [ ] STM32 board with "STM32 Boards (selected from submenu)" core
* [ ] STM32 board with "STM32F1 Boards (STM32duino.com)" core
* [ ] Apollo3 board
* [x] Teensy board
* [ ] Other - please specify

### IDE
* [ ] Arduino IDE
* [ ] Arduino Pro IDE
* [ ] Sloeber IDE
* [x] PlatformIO IDE
* [ ] Other - please specify

### IR-Protocol
* [ ] Unknown
* [x] Sony
* [ ] NEC
* [ ] Samsung
* [ ] RC5, RC6
* [ ] Kaseikyo
* [ ] Denon
* [ ] JVC
* [ ] Lego
* [ ] Matsushita
* [ ] Mitsubishi
* [x] Panasonic
* [ ] Sanyo
* [ ] Sharp
* [ ] Telefunken
* [ ] Other - please specify

### Example to reproduce the issue
* [ ] AllProtocols
* [ ] Callback
* [ ] Interrupt
* [ ] OneProtocol
* [ ] ReceiveAndSend
* [ ] ReceiveAndSendDynamicPins
* [ ] RFProtocols
* [x] SimpleReceiver
* [ ] SimpleSender
* [ ] Other

### Pin(s) used for IR-receive, if not default
22 (tried also 2 and 3)

### Version
* [x] Yes I use the [latest release](https://github.com/ukw100/IRMP/releases) and verified this!
* [ ] Other - please specify

### Current behavior
START src/main.cpp from Oct  7 2020
Using library version 3.3.2
Ready to receive IR signals of protocols: SIRCS, NEC, SAMSUNG, MATSUSH, KASEIKYO, RC5, DENON, RC6, SAMSG32, APPLE, GRUNDIG, NOKIA, SIEMENS, JVC, RC6A, IR60, NEC16, NEC42, ONKYO, at pin 22

No IR code is detected.

### Expected behavior
Should detect IR codes.

### Error output
No error.

### Additional context
Using Teensy 4.0 (Cortex-M7)

ArminJo, 2020-10-07:
Is the feedback LED flickering when IR signal is present?
Do you have a photo of your circuit?

lpaolini, 2020-10-07:
No it's not flickering.

I'm attaching a screenshot of the EAGLE schematics, if this is what you mean.
The IR sensor is connected to SV2 (1=signal, 2=+3.3v, 3=GND), while power is connected to SV1.
[image]

ArminJo, 2020-10-07:
Thanks. It looks good so far. Are you sure the IR signal is delivered at pin 22. Check the pin with static HIGH and LOW level and watch the feedback led, it should change according to the static input signal.

lpaolini, 2020-10-07:
Forgot to mention the same code runs perfectly well on Arduino Nano.

lpaolini, 2020-10-07:
Yes, I'm sure it's delivered to pin 22 (tried also 2 and 3). Tested with logic analyzer.
Nope, if I manually ground pin 22 the feedback led doesn't flicker.

ArminJo, 2020-10-07:
OK, now we have two possible reasons.
1. The program does not really use pin 22 (maybe of strange pin mappings)
2. the Interrupt routine is not running (more likely)
I do not own a Teensy 4 so I can only guess :-(
I now have a meeting for 2 hours but I after this I will do some research on it and give you feedback.

lpaolini, 2020-10-07:
Thanks.
BTW, you should own a Teensy 4, it's awesome! :)

lpaolini, 2020-10-08:
Please just let me know if there's any test I can do for you to find out the issue. Thanks.

ArminJo, 2020-10-08:
Hi, it takes longer than expected and I am very busy, so it may last until monday. Sorry.

lpaolini, 2020-10-08:
No problem. Awesome library and great support, I can wait.
Thanks.

ArminJo, 2020-10-11:
I uploaded a version with teensyduino support. Default receive pin is 3, I hope thats OK.
Can you check, if it is working?
Thanks!

lpaolini, 2020-10-12:
YES!! It works! Thank you so much for fixing it.

lpaolini, 2020-10-12:
By the way, I've tested it on pin 22 only. Please let me know if you need more tests.

ArminJo, 2020-10-12:
You would do me a big favor if you could test it on pin 3 too, since this is the default for the library. I do not know if every Teensy board has pin 22, but I assume they all have pin3 😀

lpaolini, 2020-10-12:
Sure! I confirm it's working good on default pin 3. Good job!

closed 2020-10-12 by ArminJo