IRMP-org/IRMP #26: SIEMENS protocol not detected

eku, 2020-10-20:
## Bug Report

### Arduino Platform
* [ ] Arduino ATmega328* board (UNO, Nano)
* [ ] Arduino ATmega2560 board (Mega)
* [ ] Arduino ATmega32U4 board (Leonardo)
* [ ] Arduino megaAVR board (NanoEvery)
* [ ] Arduino SAM board (Due)
* [ ] Arduino SAMD board (Zero, MKR*)
* [x] ESP8266 board
* [ ] ESP32 board
* [ ] STM32 board with "STM32 Boards (selected from submenu)" core
* [ ] STM32 board with "STM32F1 Boards (STM32duino.com)" core
* [ ] Apollo3 board
* [ ] Teensy board
* [ ] Other - please specify

### IDE
* [ ] Arduino IDE
* [ ] Arduino Pro IDE
* [ ] Sloeber IDE
* [x] PlatformIO IDE
* [ ] Other - please specify

### IR-Protocol
* [ ] Unknown
* [ ] Sony
* [ ] NEC
* [ ] Samsung
* [ ] RC5, RC6
* [ ] Kaseikyo
* [ ] Denon
* [ ] JVC
* [ ] Lego
* [ ] Matsushita
* [ ] Mitsubishi
* [ ] Panasonic
* [ ] Sanyo
* [ ] Sharp
* [ ] Telefunken
* [x] Siemens + Nokia
### Example to reproduce the issue
* [ ] AllProtocols
* [ ] Callback
* [ ] Interrupt
* [ ] OneProtocol
* [ ] ReceiveAndSend
* [ ] ReceiveAndSendDynamicPins
* [ ] RFProtocols
* [ ] SimpleReceiver
* [ ] SimpleSender
* [x] Other

### Pin(s) used for IR-receive, if not default

I use dynamic PINs (GPIO5) with Tasmota configuration.

### Version
* [x] Yes I use the [latest release](https://github.com/ukw100/IRMP/releases) and verified this!
* [ ] Other - please specify

### Current behavior
I use a chinese IR TUYA device flashed with Tasmota and my port of IRMP to Tasmota.

For comparison I have a selfmade module on with ATmega32 and native IRMP (no Arduino). This one detects all protocols correctly.

IRMP is compiled with the following settings:
```
#define IRMP_PROTOCOL_NAMES 1
#define IRMP_IRSND_ALLOW_DYNAMIC_PINS 1
#define F_INTERRUPTS 20000 // Instead of default 15000 to support LEGO + RCMM protocols
#define USE_ONE_TIMER_FOR_IRMP_AND_IRSND 1
#define tone(a,b) void() // no tone() available on ESP32
#define noTone(a) void()
#define IRMP_SUPPORT_GRUNDIG_PROTOCOL           1
#define IRMP_SUPPORT_MATSUSHITA_PROTOCOL        1
#define IRMP_SUPPORT_NEC_PROTOCOL               1
#define IRMP_SUPPORT_NOKIA_PROTOCOL             1
#define IRMP_SUPPORT_NUBERT_PROTOCOL            1
#define IRMP_SUPPORT_ONKYO_PROTOCOL             1
#define IRMP_SUPPORT_RC5_PROTOCOL               1
#define IRMP_SUPPORT_SIEMENS_PROTOCOL           1
#define IRSND_SUPPORT_GRUNDIG_PROTOCOL          1
#define IRSND_SUPPORT_MATSUSHITA_PROTOCOL       1
#define IRSND_SUPPORT_NEC_PROTOCOL              1
#define IRSND_SUPPORT_NOKIA_PROTOCOL            1
#define IRSND_SUPPORT_NUBERT_PROTOCOL           1
#define IRSND_SUPPORT_ONKYO_PROTOCOL            1
#define IRSND_SUPPORT_RC5_PROTOCOL              1
#define IRSND_SUPPORT_SIEMENS_PROTOCOL          1
```

### Expected behavior
Siemens and Nokia protocols are always correctly detected.

### Error output
No serial output on Tasmota :(

The Siemens protocol is  not detected at all.
The Nokia protocol is sometimes detected as SIEMENS.

### Additional context

ArminJo, 2020-10-22:
First it can be a problem of the IR receiver hardware.
Second, I do not know if Tasmota interferes with the polling of IRMP.
What about other protocols e.g. NEC?

eku, 2020-10-23:
I have reduced the sampling frequency to 15kHz as a test and Nokia is reliably detected. Siemens is at least sometimes detected, but not deterministically.

As receiver I use the same TSOP in both circuits.

The only thing that remains, as you already suspect, is the different implementation of the timers. Regarding the Arduino implementation for ESP, is there anything else you can think of where you can pull the lever and what else you could try?

ArminJo, 2020-10-23:
How can I reproduce it? I have ESP's but only worked once with platformio and never with tasmota.
Are there some tutorials, or can I copy your platformio project. 
Maybe some debugs can be enabled in IRMP to get an idea why it fails for Siemens.

eku, 2020-10-23:
Use my branch: https://github.com/eku/Tasmota/tree/feature/irmp_support
and edit https://github.com/eku/Tasmota/blob/feature/irmp_support/tasmota/xdrv_45_irmp.ino as needed.
In addition, you need to activate the feature in https://github.com/eku/Tasmota/blob/feature/irmp_support/tasmota/tasmota_configurations.h (USE_IRMP).

ArminJo, 2020-11-26:
See also https://www.mikrocontroller.net/topic/goto_post/6483340

eku, 2021-02-03:
What is the next step with this topic? Has @ukw100 made an bug fix?

ArminJo, 2021-02-03:
I commited timing the corrections.

eku, 2021-02-15:
Version 3.4.0 does not improve the recognition.

eku, 2022-03-08:
Version 3.5.1 doesn't decode SIEMENS at all.

eku, 2022-03-12:
I did upgrade to 1a62365c187c6cac2e0f80463952db3eb930457a, but no luck. 

I don't know if it is necessarily due to the timing. I also use IRMP on an ATmega, but not the ARDUNIO version, but integrated in my own software and SIEMENS, RUWIDO and NOKIA are recognised. Could the ESP8266 cause more problems here than the AVR?

The recognition rate is rather worse at 3.60.

I have both a real SIEMENS remote and an universal remote that learned the codes from it. UKW originally implemented the SIEMENS protocol first based on my recorded scans. I use the same nick on mikrocontroller.net and I'm familiar with IRMP since years.

ArminJo, 2022-03-12:
Until the last change I used Franks IRMP timing. And in the forum, he recommends the new timings of https://github.com/j1rie/IRMP. And of course I did not touch the logic!
So it must be the ESP8266 hardware (or timer) which makes the difference.