IRMP-org/IRMP #27: ATmega8A, any chance?

SanZamoyski, 2020-10-27:
## Bug Report

### Arduino Platform
* [ ] Other - please specify: atmega8a barebone

### IDE
* [ ] Arduino IDE

### IR-Protocol
* [ ] Kaseikyo

### Example to reproduce the issue
* [ ] Other

### Pin(s) used for IR-receive, if not default

### Version
* [ ] Yes I use the [latest release](https://github.com/ukw100/IRMP/releases) and verified this!
* [ ] Other - please specify

### Current behavior
I'm trying to move my sketch from Arduino nano onto barebone 8A, on pages and in the code it is sometimes mentioned that this should work on 8a. I believe it was but it does not now.

I realised that there are some convention on that, further I've found it is sometime used as __AVR_ATmega8A__, so I tried to modify sources:

```
san@lati:~/Pobrane/IRMP-master$ diff --suppress-common-lines -y -r vanilla/ modified
diff --suppress-common-lines -y -r vanilla/src/irsndconfig.h modified/src/irsndconfig.h
#  define IRSND_OCx                             IRSND_OC2B    |
                                                              > #  if defined(__AVR_ATmega8A__)
                                                              > #    define IRSND_OCx                             IRSND_OC2
                                                              > #  else
                                                              > #    define IRSND_OCx                             IRSND_OC2B 
                                                              > #  endif
diff --suppress-common-lines -y -r vanilla/src/IRTimer.cpp.h modified/src/IRTimer.cpp.h
#  if defined(__AVR_ATmega16__)                               | #  if defined(__AVR_ATmega8A__) || defined(__AVR_ATmega16__)
#  if defined(__AVR_ATmega16__)                               |     
                                                              > #  elif defined(__AVR_ATmega8A__) || defined(__AVR_ATmega16__
                                                              |     
#  if defined(__AVR_ATmega16__)                               | #  if defined(__AVR_ATmega16__) || defined(__AVR_ATmega8A__)
                                                              > #  elif defined(__AVR_ATmega8A__)
                                                              >     TIMSK = 0; // disable interrupt
#  if defined(__AVR_ATmega16__)                               | #  if defined(__AVR_ATmega16__) || defined(__AVR_ATmega8A__)

```

but, ofcourse it does not work.

I've tried to debug it somehow by this code:
/*
 * Set library modifiers first to set output pin etc.
 */
//#include "PinDefinitionsAndMore.h"
#define __AVR_ATmega8A__

#define IRSND_IR_FREQUENCY          38000
#define IRSND_OUTPUT_PIN 3

#define IRSND_SELECT_PROTOCOLS_H
#define IRSND_SUPPORT_KASEIKYO_PROTOCOL         1       // Kaseikyo             >= 10000                 ~300 bytes

```
/*
 * After setting the definitions we can include the code and compile it.
 */
#include <irsnd.c.h>

IRMP_DATA pan_power, pan_input, pan_volup, pan_voldown;

#include <avr/sleep.h>

const int ledIR = 2;

void setup() {
    
  pinMode(input, INPUT_PULLUP);
  irsnd_init();

  //define data packs for 4 ir commands
  pan_power.protocol = IRMP_KASEIKYO_PROTOCOL;
  pan_power.address = 0x2002;
  pan_power.command = 0x83D0;
  pan_power.flags = 10;  
  
  Serial.println("Blink!");
  digitalWrite(ledIR, HIGH);
  delay(1000);
  digitalWrite(ledIR, LOW);
  delay(1000);
  
  irsnd_send_data(&pan_power, true);
  
  Serial.println("Blink!");
  digitalWrite(ledIR, HIGH);
  delay(1000);
  digitalWrite(ledIR, LOW);
  delay(1000);
}
```
(i've removed som irrelevant stuff)

ATmega blinks once only, so it has to hang on irsnd_send_data(). I have no idea what to do next. I'm trying to compare ATmega16 and 8a datasheets, but no luck.

ArminJo, 2020-10-27:
Which Arduino boards Manager URL do you use?

SanZamoyski, 2020-10-27:
```
https://mcudude.github.io/MicroCore/package_MCUdude_MicroCore_index.json
https://raw.githubusercontent.com/damellis/attiny/ide-1.6.x-boards-manager/package_damellis_attiny_index.json
https://raw.githubusercontent.com/carlosefr/atmega/master/package_carlosefr_atmega_index.json
```

is what You ask?

SanZamoyski, 2020-10-27:
I've changed it to:
```
https://mcudude.github.io/MicroCore/package_MCUdude_MicroCore_index.json
https://mcudude.github.io/MiniCore/package_MCUdude_MiniCore_index.json
```
But still it hangs.

SanZamoyski, 2020-10-28:
That's odd, IRremote library uses different pin and timer for tx:

```
// Arduino Duemilanove, Diecimila, LilyPad, Mini, Fio, Nano, etc
// ATmega48, ATmega88, ATmega168, ATmega328
#elif defined(__AVR_ATmega328P__) || defined(__AVR_ATmega328PB__) || defined(__AVR_ATmega168__) // old default clause
#  if !defined(IR_USE_TIMER1) && !defined(IR_USE_TIMER2)
//#define IR_USE_TIMER1   // tx = pin 9
#define IR_USE_TIMER2     // tx = pin 3
#  endif

// Atmega8
#elif defined(__AVR_ATmega8__)
#  if !defined(IR_USE_TIMER1)
#define IR_USE_TIMER1     // tx = pin 9
#  endif

// MightyCore - ATmega8535, ATmega16, ATmega32
#elif defined(__AVR_ATmega8535__) || defined(__AVR_ATmega16__) || defined(__AVR_ATmega32__)
#  if !defined(IR_USE_TIMER1)
#define IR_USE_TIMER1     // tx = pin 13
#  endif
```

from: https://github.com/z3t0/Arduino-IRremote/blob/master/src/private/IRremoteBoardDefs.h

ArminJo, 2020-10-28:
I changed the IRTimer.cpp.h to support ATMega8
Just use the current version from the repo.

SanZamoyski, 2020-10-28:
Hi ArminJo,

many thanks, it does not hang now. I had some problems with sleep function (didn't know external interrupt wake up in atmega8a does not work for FALLING, only for LOW and HIGH), that's why I'm answering that late. But now my sketch almost works.

Almost because:
```
arduino_modified_sketch_1629/SimpleReceiver.ino from Oct 26 2020␍␊
[16:51:57:379] Using library version 3.3.2␍␊
[16:51:57:382] Ready to receive IR signals of protocols: SIRCS, NEC, SAMSUNG, MATSUSH, KASEIKYO, RC5, DENON, RC6, SAMSG32, APPLE, GRUNDIG, NOKIA, SIEMENS, JVC, RC6A, IR60, NEC16, NEC42, ONKYO, at pin 11␍␊
```
Receives nothing.
To check this I've uploaded SimpleSender with `#define IRSND_OUTPUT_PIN 3`. But still nothing. Other remote (NEC protocol) works.

Best regards!

ArminJo, 2020-10-28:
Is the value of F_CPU during compile 8000000?

SanZamoyski, 2020-10-28:
I've added 
```
    Serial.print("F_CPU: ");
    Serial.print(F_CPU);
```
after `Serial.begin` and it prints 8000000.

ArminJo, 2020-10-28:
Then you have done al right.
Is the receiver led on the receiving arduino blinking (like with NEC) if you send with the ATmega8?

SanZamoyski, 2020-10-28:
You mean one at D13? 
Yes and as long as i can see - the same pattern as sender.

SanZamoyski, 2020-10-28:
It receive:

```
[20:06:21:991] P=SIEMENS  A=0x5AA C=0x12A␍␊
[20:06:25:149] P=SIEMENS  A=0x3D5 C=0x15A␍␊
[20:06:26:928] P=  A=0xAB C=0x2A␍␊
[20:06:27:997] P=SIEMENS  A=0x36B C=0x5A␍␊
[20:06:29:272] P=SIEMENS  A=0x16B C=0x2DA␍␊
[20:06:32:059] P=  A=0x6B C=0x56␍␊
[20:06:37:902] P=SIEMENS  A=0x5EA C=0x15A␍␊
[20:06:38:662] P=SIEMENS  A=0x2A9 C=0xDA␍␊
[20:06:39:008] P=SIEMENS  A=0x2BA C=0x352␍␊
[20:06:48:003] P=  A=0xBC C=0x56␍␊
[20:06:58:566] P=  A=0xDB C=0x6A␍␊
[20:06:59:993] P=SIEMENS  A=0x2BB C=0x156␍␊
[20:07:04:262] P=  A=0xD5 C=0x2A␍␊
[20:07:09:899] P=SIEMENS  A=0x355 C=0x172␍␊
[20:07:11:119] P=  A=0x8A C=0x4A␍␊
[20:07:19:751] P=SIEMENS  A=0x25A C=0x2B6␍␊
[20:07:19:899] P=  A=0xB6 C=0x56␍␊
[20:07:21:783] P=SIEMENS  A=0x2DD C=0x5A␍␊
[20:07:26:409] P=  A=0xAA C=0x6A␍␊
[20:07:26:612] P=SIEMENS  A=0x2AA C=0x1DA␍␊
```

SanZamoyski, 2020-10-28:
I believe that this can be closed. I drive ir led wrong. I need to build some additional circuit, but don't have parts at the moment. That's propably the case.

Once again, thank You for Your work and fast adding 8a chip to supported devices.

Best regards!

closed 2020-10-28 by SanZamoyski

ArminJo, 2020-10-28:
I connected my IR diode with a 270 Ohm resistor and everything works well. 
But you now can easily test it by getting again the latest version from the repo and defining `IRSND_GENERATE_NO_SEND_RF` in the sender sketch at the very beginning and then disconnect the sender IR diode and connect the sender output pin to the receiver input pin (and connect ground of the two boards also).

SanZamoyski, 2020-11-01:
Unfortunately, I still have a problem.
Using NO_SEND_RF gives good results, but using real diode (through 220R) gives nothing correct.
I'm using TSAL6100 which is everywhere considered as best option. Although LED on receiver blinks...

ArminJo, 2020-11-02:
This looks like a receiver circuit problem like [here ](https://github.com/z3t0/Arduino-IRremote/issues/686)and sone from [here](https://github.com/z3t0/Arduino-IRremote/issues?q=is%3Aissue+receiver).

SanZamoyski, 2020-11-02:
I can not be receiver: TV does not respond to any code also I've recorded those codes with this receiver.

ArminJo, 2020-11-02:
Maybe your real CPU frequency is slightly off, so the modulation is not 38 kHz. 
Is sending NEC still possible with the new settings?

SanZamoyski, 2020-11-02:
According to what datasheet says:
```
During reset, hardware loads the 1MHz calibration byte into the OSCCAL Register and thereby automatically calibrates the RC Oscillator. At 5V, 25C and 1.0MHz Oscillator frequency
selected, this calibration gives a frequency within ± 3% of the nominal frequency.
```
Although I've read that it can be up to 10% (but it was maybe across temperature range). 
Is 3% too much, is 10% too much?

Oh, I'm testing send-receive with samples sketch. After I realised my program does not work (just to be sure that nothing else causes that.

SanZamoyski, 2020-11-03:
BUT!

If the modulatiom is off, is it possible that sketch with NO_SEND_RF works?

ArminJo, 2020-11-03:
I am totally lost. 
Please make a list including what you have tested and the test results.

The [IRSND_GENERATE_NO_SEND_RF](https://github.com/ukw100/IRMP#compile-options--macros-for-this-library) is for back to back (connecting by wire) testing of sender and receiver.

SanZamoyski, 2020-11-04:
OK!
Everything was done with ATmega8A (internal 8 MHz, TSAL6100 with 220R, SimpleSender sketch) as _sender_ and Arduino Nano clone as _receiver_ (1838B as infrared module - connected AS IS, SimpleReceiver sketch).

1. Sender and receiver connected via wire (no IR diode, no 1838B). Everything works just fine.
2. _sender_ (with my own sketch) to TV - no response.
3. _sender_ to _receiver_ - no response.

ArminJo, 2020-11-04:
> sender to receiver - no response.

But feedback led blinks???

what are the results for sender to receiver with sending NEC (this is foolproof)?

SanZamoyski, 2020-11-04:
> But feedback led blinks???
Yes.

> what are the results for sender to receiver with sending NEC (this is foolproof)?
By "sending NEC" do You mean SimpleSender? It does write it is ready to receive, but no further messages.

ArminJo, 2020-11-04:
Simple sender does not write ready to receive, maybe this is the bug.