IRMP-org/IRMP #28: [Question] Use 2 or more IR leds to send

onolox, 2020-11-07:
* [ ] Arduino ATmega328* board (UNO, Nano)

### IDE
* [ ] PlatformIO IDE

### IR-Protocol
* [ ] NEC

### Example to reproduce the issue
* [ ] AllProtocols

### Version
* [ ] Yes I use the [latest release](https://github.com/ukw100/IRMP/releases) and verified this!

### Expected behavior

Hi, I want to use 2 IR leds to send to 2 different devices that are at different locations, I don't understant the example of IRMP_IRSND_ALLOW_DYNAMIC_PINS. 

How I can put the 2 leds in 2 pins of arduino?

ArminJo, 2020-11-10:
Use the current clone pattern known as series connection.
Do not forget to reduce the series resistor to get the same current as with 1 LED.
And then use just one pin, it will work!

onolox, 2020-11-10:
I know that I can use this approach, but i want a more elegant solution.

ArminJo, 2020-11-10:
Call irsnd_init(tIRSNDOutputPin) each time you switch the output pin eg. before sending.
I have not tested it, but it should work.

closed 2020-11-15 by ArminJo