IRMP-org/IRMP #31: Timer problems on Leonardo

tobiasdiez, 2020-11-21:
## Bug Report
Compiling the SimpleReceiver example yields the following error message:
```
n file included from D:\Documents\Arduino\libraries\IRMP\src/irmpArduinoExt.cpp.h:18:0,
                 from D:\Documents\Arduino\libraries\IRMP\src/irmp.c.h:3143,
                 from D:\Programming\Projects\Arduino\receive-ir\receive-ir.ino:65:
D:\Documents\Arduino\libraries\IRMP\src/IRTimer.cpp.h: In function 'void initIRTimerForReceive()':
D:\Documents\Arduino\libraries\IRMP\src/IRTimer.cpp.h:182:5: error: 'TCCR2A' was not declared in this scope
     TCCR2A = _BV(WGM21); // CTC mode
     ^~~~~~
D:\Documents\Arduino\libraries\IRMP\src/IRTimer.cpp.h:182:5: note: suggested alternative: 'TCCR0A'
     TCCR2A = _BV(WGM21); // CTC mode
     ^~~~~~
     TCCR0A
In file included from c:\program files (x86)\arduino\hardware\tools\avr\avr\include\avr\io.h:99:0,
                 from c:\program files (x86)\arduino\hardware\tools\avr\avr\include\avr\pgmspace.h:90,
                 from C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:28,
                 from D:\Programming\Projects\Arduino\receive-ir\receive-ir.ino:47:
D:\Documents\Arduino\libraries\IRMP\src/IRTimer.cpp.h:182:18: error: 'WGM21' was not declared in this scope
     TCCR2A = _BV(WGM21); // CTC mode
                  ^
D:\Documents\Arduino\libraries\IRMP\src/IRTimer.cpp.h:182:18: note: suggested alternative: 'WGM41'
In file included from D:\Documents\Arduino\libraries\IRMP\src/irmpArduinoExt.cpp.h:18:0,
                 from D:\Documents\Arduino\libraries\IRMP\src/irmp.c.h:3143,
                 from D:\Programming\Projects\Arduino\receive-ir\receive-ir.ino:65:
D:\Documents\Arduino\libraries\IRMP\src/IRTimer.cpp.h:187:5: error: 'TCCR2B' was not declared in this scope
     TCCR2B = _BV(CS21);                                             // prescale by 8
     ^~~~~~
D:\Documents\Arduino\libraries\IRMP\src/IRTimer.cpp.h:187:5: note: suggested alternative: 'TCCR0B'
     TCCR2B = _BV(CS21);                                             // prescale by 8
     ^~~~~~
     TCCR0B
In file included from c:\program files (x86)\arduino\hardware\tools\avr\avr\include\avr\io.h:99:0,
                 from c:\program files (x86)\arduino\hardware\tools\avr\avr\include\avr\pgmspace.h:90,
                 from C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:28,
                 from D:\Programming\Projects\Arduino\receive-ir\receive-ir.ino:47:
D:\Documents\Arduino\libraries\IRMP\src/IRTimer.cpp.h:187:18: error: 'CS21' was not declared in this scope
     TCCR2B = _BV(CS21);                                             // prescale by 8
                  ^
D:\Documents\Arduino\libraries\IRMP\src/IRTimer.cpp.h:187:18: note: suggested alternative: 'CS01'
In file included from D:\Documents\Arduino\libraries\IRMP\src/irmpArduinoExt.cpp.h:18:0,
                 from D:\Documents\Arduino\libraries\IRMP\src/irmp.c.h:3143,
                 from D:\Programming\Projects\Arduino\receive-ir\receive-ir.ino:65:
D:\Documents\Arduino\libraries\IRMP\src/IRTimer.cpp.h:188:5: error: 'OCR2B' was not declared in this scope
     OCR2B = OCR2A = ((F_CPU / 8) / IR_INTERRUPT_FREQUENCY) - 1;     // 132 for 15000 interrupts per second
     ^~~~~
D:\Documents\Arduino\libraries\IRMP\src/IRTimer.cpp.h:188:5: note: suggested alternative: 'OCR4B'
     OCR2B = OCR2A = ((F_CPU / 8) / IR_INTERRUPT_FREQUENCY) - 1;     // 132 for 15000 interrupts per second
     ^~~~~
     OCR4B
D:\Documents\Arduino\libraries\IRMP\src/IRTimer.cpp.h:188:13: error: 'OCR2A' was not declared in this scope
     OCR2B = OCR2A = ((F_CPU / 8) / IR_INTERRUPT_FREQUENCY) - 1;     // 132 for 15000 interrupts per second
             ^~~~~
D:\Documents\Arduino\libraries\IRMP\src/IRTimer.cpp.h:188:13: note: suggested alternative: 'OCR3A'
     OCR2B = OCR2A = ((F_CPU / 8) / IR_INTERRUPT_FREQUENCY) - 1;     // 132 for 15000 interrupts per second
             ^~~~~
             OCR3A
D:\Documents\Arduino\libraries\IRMP\src/IRTimer.cpp.h:190:5: error: 'TIFR2' was not declared in this scope
     TIFR2 = _BV(OCF2B) | _BV(OCF2A) | _BV(TOV2);                    // reset interrupt flags
     ^~~~~
D:\Documents\Arduino\libraries\IRMP\src/IRTimer.cpp.h:190:5: note: suggested alternative: 'TIFR4'
     TIFR2 = _BV(OCF2B) | _BV(OCF2A) | _BV(TOV2);                    // reset interrupt flags
     ^~~~~
     TIFR4
In file included from c:\program files (x86)\arduino\hardware\tools\avr\avr\include\avr\io.h:99:0,
                 from c:\program files (x86)\arduino\hardware\tools\avr\avr\include\avr\pgmspace.h:90,
                 from C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:28,
                 from D:\Programming\Projects\Arduino\receive-ir\receive-ir.ino:47:
D:\Documents\Arduino\libraries\IRMP\src/IRTimer.cpp.h:190:17: error: 'OCF2B' was not declared in this scope
     TIFR2 = _BV(OCF2B) | _BV(OCF2A) | _BV(TOV2);                    // reset interrupt flags
[... 111 lines cut ...]
### Example to reproduce the issue
* [ ] AllProtocols
* [ ] Callback
* [ ] Interrupt
* [ ] OneProtocol
* [ ] ReceiveAndSend
* [ ] ReceiveAndSendDynamicPins
* [ ] RFProtocols
* [x] SimpleReceiver
* [ ] SimpleSender
* [ ] Other

### Pin(s) used for IR-receive, if not default

### Version
* [x] Yes I use the [latest release](https://github.com/ukw100/IRMP/releases) and verified this!
* [ ] Other - please specify

### Current behavior

### Expected behavior

### Error output

### Additional context

ArminJo, 2020-11-21:
UUUps. and Sorry.
Leonardo Support was missing.
Please try the new version.
Thanks for reporting!

closed 2020-11-21 by ArminJo

ArminJo, 2020-11-21:
And now tested. 😀

commit b084a861 (2020-11-21): Added missing Leonardo support #closes #31
commit 88765c38 (2020-11-21): Added missing Leonardo support #closes #31
commit cbd25700 (2020-11-21): Added missing Leonardo support #closes #31