IRMP-org/IRMP #34: Redefinitions causing code to fail compilation on Sparkfun Artemis Boards (Apollo 3)

Marsfan, 2020-12-27:
## Bug Report

### Arduino Platform
* [x] Apollo3 board

### IDE
* [x] Arduino IDE

### IR-Protocol
* [x] Other - Not relavent

### Example to reproduce the issue
* [x] AllProtocols
* [x] SendAllProtocols 
* [ ] Callback
* [ ] Interrupt
* [ ] OneProtocol
* [ ] ReceiveAndSend
* [ ] ReceiveAndSendDynamicPins
* [ ] RFProtocols
* [ ] SimpleReceiver
* [ ] SimpleSender
* [ ] Other

### Pin(s) used for IR-receive, if not default

### Version
* [x] Yes I use the [latest release](https://github.com/ukw100/IRMP/releases) and verified this!
* [ ] Other - please specify

### Current behavior

Code is the AllProtocols example with no changes made to it

<details>
<summary>Output</summary>

```
In file included from D:\Documents\Arduino\libraries\IRMP\src/irmp.h:31,
                 from D:\Documents\Arduino\libraries\IRMP\src/irmp.c.h:26,
                 from C:\Users\Gabe\AppData\Local\Temp\arduino_modified_sketch_266082\AllProtocols.ino:88:
D:\Documents\Arduino\libraries\IRMP\src/irmpArduinoExt.h:84: warning: "F" redefined
 #define F(a) a
 
In file included from C:\Users\Gabe\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.0.3/cores/arduino/mbed-bridge/core-api/api/Print.h:24,
                 from C:\Users\Gabe\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.0.3/cores/arduino/mbed-bridge/core-api/api/Stream.h:25,
                 from C:\Users\Gabe\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.0.3/cores/arduino/mbed-bridge/core-api/api/Client.h:22,
                 from C:\Users\Gabe\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.0.3/cores/arduino/mbed-bridge/core-api/api/ArduinoAPI.h:29,
                 from C:\Users\Gabe\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.0.3/cores/arduino/mbed-bridge/Arduino.h:14,
                 from C:\Users\Gabe\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.0.3/cores/arduino/sdk/ArduinoSDK.h:9,
                 from <command-line>:
C:\Users\Gabe\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.0.3/cores/arduino/mbed-bridge/core-api/api/String.h:45: note: this is the location of the previous definition
 #define F(string_literal) (reinterpret_cast<const __FlashStringHelper *>(PSTR(string_literal)))
 
c:/users/gabe/appdata/local/arduino15/packages/sparkfun/tools/arm-none-eabi-gcc/8-2018-q4-major/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\Gabe\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.0.3\variants\SFE_ARTEMIS_ATP/mbed/libmbed-os.a(isr.o): in function `am_ctimer_isr':
/github/workspace/mbed-builds/artemis-redboard-atp-lib_1/mbed-os/targets/TARGET_Ambiq_Micro/TARGET_Apollo3/device/isr.c:6: multiple definition of `am_ctimer_isr'; sketch\AllProtocols.ino.cpp.o:D:\Documents\Arduino\libraries\IRMP\src/IRTimer.cpp.h:657: first defined here
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board RedBoard Artemis ATP.

```
</details>

Code fails during compilation due to redefinition of `am_ctimer_isr(void)` and `F(a)`

### Expected behavior

Code Compiles correctly. 

### Additional context

It seems that Sparkfun added their own functions that implement the interrupts and `F(a)` substitution. 
* **Interrupt** function:  Identical to the interrupt function that that this repository has implemented.
* **Flash String Helper**: Differs from this repo's implementation by using the true Arduino implementation. 

I tested commenting out the function definitions of `am_ctimer_isr(void)` and `F(a)` in this library and found that the Arduino IDE properly compiles the `AllProtocols` example when the configured board is the Sparkfun Artemis Redboard ATP. I will plan to verify that the library works properly with the code commented out, and will submit a PR if I find that it does.

ArminJo, 2020-12-27:
Hi, I am not able to reporduce the error I only get some strange BFD errors at the end of linking.

Marsfan, 2020-12-28:
@ArminJo BFD Error on the Sparkfun Artemis? I don't even know what that is. I have a little bit of experience with wading through the Artemis docs, can you upload your error output?

ArminJo, 2020-12-28:
Here they are: https://github.com/ukw100/IRMP/runs/1615150246?check_suite_focus=true#step:3:622 
search for BFD and you will find a lot. I get the same error on W10 with the Arduino IDE, so support is highly appreciated 👍

ArminJo, 2020-12-28:
The BFD error is fixed! 😀

Marsfan, 2020-12-28:
@ArminJo I don't know about the BFD error, but I can tell you this, because i just fixed this myself earlier today. You are using a really old version of the Artemis Board Package (You are using 1.2.1, newest is 2.0.3). They changed the board manager URL to [https://raw.githubusercontent.com/sparkfun/Arduino_Apollo3/master/package_sparkfun_apollo3_index.json](https://raw.githubusercontent.com/sparkfun/Arduino_Apollo3/master/package_sparkfun_apollo3_index.json). 

I bet that updating that will also cause you to have the same error that I opened this bug report for. Like I said, I just realized and fixed the URL myself earlier today.

ArminJo, 2020-12-28:
Thanks for the new board manager URL!
I changed the sources now, so that no errors should happen any more.
Can you please do me the favor and check if the timer works like before? I.e. IR receiving is working and the feedback led is flickering.

Marsfan, 2020-12-28:
@ArminJo I was working on that earlier and plugged my IR receiver in backwards and destroyed it, so I cannot test it right now. 
I likely will not be able to get another IR receiver until January 6, but I will look around and see if I have one laying around that I previously did not find. 
I am currently working on writing a script to run a test of the transmission mode, so I hope to have done tonight though.

Marsfan, 2020-12-30:
@ArminJo I found another IR receiver and tested it. It now compiles fine and runs fine. on the Artemis ATP

closed 2020-12-30 by ArminJo

Marsfan, 2021-01-05:
@ArminJo I just was looking around. You might actually want to test against both V1.2.1 cores and V2.x cores for the Artemis. Some of the capabilities from the V1 core are still not available in the V2 core, so some people still use the V1 core.

ArminJo, 2021-01-05:
Thanks, for the advice 👍

PR #36 (2020-12-30): Removed no longer needed lines for Apollo 3
This fixes #34. I have run both simpleSend and simpleReceive to verify that it works properly on a board that uses the Sparkfun Apollo3 as its core.