IRMP-org/IRMP #35: ESP32; irsnd_send_data(...)

supaplextor, 2020-12-29:
## Bug Report

I'm a bit lost comparing schematics to see/know if the IR unit is right for this. Eventually I want to disconnect the attached IR unit, and use the built in (pin 9) IR from the base unit (which is an M5Stack M5StickC using esp32 core). RX seems okay. Serial monitor implies IR was sent, but device does not respond (OEM remote is fine).

I have two sets, one running IRMP, the second base/unit pair is counting pulses each second.

The second unit reads 34/34 high/low with a single OEM remote button press (stats refresh/clear each second). Zero activity while aiming retransmit from unit one (code below) to unit two (code MIA kinda).

### Arduino Platform
* [X] ESP32 class
* base: https://docs.m5stack.com/#/en/core/m5stickc
* IR on grove port: https://docs.m5stack.com/#/en/unit/ir

Willing to test on breadboard with a leaner esp32 dev board connected to IR unit above.

### IDE
* [X] Arduino IDE

### IR-Protocol
* [X] NEC (primary issue; Oon Soundbar IR remote right now)
* [X] Samsung (maybe later;have a remote on hand)

### Example to reproduce the issue
* [X] ReceiveAndSend

### Pin(s) used for IR-receive, if not default

33 RX, 9 or 32 TX, 10 red LED.

<pre>START foo/m5-ReceiveAndSend-retry3wip.ino from Dec 29 2020 07:14:08 Using library version 3.3.4
Ready to receive IR signals of protocols: SIRCS, NEC, SAMSUNG, MATSUSH, KASEIKYO, RECS80, RC5, DENON, RC6, SAMSG32, APPLE, RECS80EX, NUBERT, BANG OLU, GRUNDIG, NOKIA, SIEMENS, FDC, RCCAR, JVC, RC6A, NIKON, IR60, KATHREIN, NETBOX, NEC16, NEC42, THOMSON, BOSE, A1TVBOX, TELEFUNKEN, SPEAKER, LGAIR, SAMSG48, MERLIN, PENTAX, TECHNICS, MITSU_HEAVY, VINCENT, SAMSUNGAH, GREE, METZ, ONKYO, at pin 33
Ready to send IR signals at pin 9
setup() done.
P=NEC  A=0x7F80 C=0x12
key detected, 0x00000012
</pre>

I have a secondary pair (m5+IR unit) counting IR pulses per minute, with 34 high and 34 low states seen with the same remote button press.

### Version
* [X] Yes I use the [latest release](https://github.com/ukw100/IRMP/releases) and verified this!
* [ ] Other - please specify

*  "version": "3.3.4",

### Current behavior

Trying to abuse this call to irsnd_send_data with the object irmp_get_data used to retransmit (button click on m5/esp32).
<pre>#elif defined(ESP32)
#define IRMP_INPUT_PIN   33
#define IRSND_OUTPUT_PIN 9 // 32
#define FEEDBACK_LED_IS_ACTIVE_LOW
#define ALTERNATIVE_IRMP_FEEDBACK_LED_PIN 10;
#define tone(a,b) void() // no tone() available on ESP32
#define noTone(a) void()
#define TONE_PIN 42 // Dummy for examples using it
</pre>

<pre>#include &lt;M5StickC.h&gt;
#include &lt;Arduino.h&gt;

/*
   Set library modifiers first to set input and output pin etc.
*/
#include "PinDefinitionsAndMore.h"
#include &lt;irmpSelectAllProtocols.h&gt;  // This enables all possible protocols

#define IRSND_IR_FREQUENCY          38000

#define IRMP_PROTOCOL_NAMES         1 // Enable protocol number mapping to protocol strings - requires some FLASH.
#define IRSND_PROTOCOL_NAMES        1 // Enable protocol number mapping to protocol strings - requires some FLASH.

#define IRMP_SUPPORT_SAMSUNG_PROTOCOL     1
#define IRSND_SUPPORT_SAMSUNG_PROTOCOL    0

/*
   After setting the definitions we can include the code and compile it.
*/

#define USE_ONE_TIMER_FOR_IRMP_AND_IRSND // otherwise we get an error on AVR platform: redefinition of 'void __vector_8()
#include <irmp.c.h>
#include <irsnd.c.h>

IRMP_DATA irmp_data;
IRMP_DATA irsnd_data;

void sendSamsungSmartHubMacro(bool aDoSelect);
void IRSendWithDelay(uint16_t aCommand, uint16_t aDelayMillis);
void IRre_xmit();

void setup()
{
#if defined(MCUSR)
  MCUSR = 0; // To reset old boot flags for next boot
#endif

  Serial.begin(115200);
  delay(2000);
  M5.begin();
  Serial.println("START " __FILE__ " from " __DATE__ " " __TIME__ " Using library version " VERSION_IRMP);

  irmp_init();  
  irmp_irsnd_LEDFeedback(true); // Enable receive signal feedback at LED_BUILTIN for receive and send  
  irsnd_init();

  Serial.print(F("Ready to receive IR signals of protocols: "));
  irmp_print_active_protocols(&Serial);
  Serial.println(F("at pin " STR(IRMP_INPUT_PIN)));
  Serial.println(F("Ready to send IR signals at pin " STR(IRSND_OUTPUT_PIN)));
  Serial.println("setup() done.");
}

void loop()
{
  yield();
  if (irmp_get_data(&irmp_data)) {
    irmp_result_print(&irmp_data);
    Serial.printf("key detected, 0x%08X", irmp_data.command);
    Serial.println();
    // Flush repeats received
    while (irmp_get_data(&irmp_data))
      yield();
  }

  delay(20);
  if (M5.BtnA.wasPressed()) { // aka after button A is released.
    yield();
    IRre_xmit();

    /*
    Serial.printf("Sending 0x%08X",irmp_data.command); Serial.println();

    irsnd_data.protocol   = irmp_data.protocol;
    irsnd_data.address    = irmp_data.address;
    irsnd_data.command    = irmp_data.command;
    irsnd_data.flags      = irmp_data.flags;
    
    IRSendWithDelay(irsnd_data.command, 0);
    */
    
    Serial.println("Sent.");
  }

  if (M5.BtnB.wasPressed()) {
    yield();
    irmp_data.command = int(random(0x100));
//    Serial.printf("Sending 0x%08X",irmp_data.command); Serial.println();

    irsnd_data.protocol   = irmp_data.protocol;
    irsnd_data.address    = irmp_data.address;
    irsnd_data.command    = irmp_data.command;
    irsnd_data.flags      = irmp_data.flags;
    
    IRSendWithDelay(irsnd_data.command, 0);
    // Serial.println("Sent.");
    
  }

  M5.update();
}

void IRre_xmit()
{
  
  irsnd_send_data(&irmp_data, true); // true = wait for frame to end. This stores timer state and restores it after sending
  irsnd_data_print(&Serial, &irmp_data);
}

void IRSendWithDelay(uint16_t aCommand, uint16_t aDelayMillis)
{
  irsnd_data.command = aCommand;      // For my Samsung, the high byte is the inverse of the low byte, this is not checked here.
  irsnd_send_data(&irsnd_data, true); // true = wait for frame to end. This stores timer state and restores it after sending
  irsnd_data_print(&Serial, &irsnd_data);
  delay(aDelayMillis);
}</pre>

supaplextor, 2020-12-29:
[image]

ArminJo, 2020-12-29:
Did you test the SimpleReceiver, the SimpleSender and AllProtocols?
What are the test results?
Why do you specify `#define IRSND_SUPPORT_SAMSUNG_PROTOCOL    0` ?

supaplextor, 2020-12-29:
SimpleReceiver, SimpleSender and AllProtocols appear to work.

output is not Samsung. Protocol=NEC  Address=0x7F80 Command=0x12
I saw the include for irsndSelectMain15Protocols.h and added it above irsnd.c.h, and now I have a working LearningRemote sample.

[m5-IRMP-LearningRemote.zip](https://github.com/ukw100/IRMP/files/5752279/m5-IRMP-LearningRemote.zip)

Got this working :)

closed 2020-12-29 by supaplextor