IRMP-org/IRMP #37: Sending IR codes without modulation / carrier frequency

dennisfrett, 2021-01-04:
I have a project in which I have a few outputs: IR LEDs and a 3.5mm wire going directly into the IR receiver of another device.

Controlling devices over the IR LEDs works fine, but to control the device over the wire I need to be able to send NEC pulses without modulation (no carrier frequency) and inverted logic.

From the documentation I understand that the macro `IRSND_GENERATE_NO_SEND_RF` can be used to send unmodulated data, but this would impact all the pulses I send. 

Is there a dynamic way to switch between sending modulated / unmodulated signals? If not, I would be open to having a look at creating a pull request that adds an API for this if other people are also interested.

ArminJo, 2021-01-04:
Hi Dennis,
thanks for the offer 👍 , but it is a rather seldom required functionality to have it switched dynamically, so we do not plan to implement it.

dennisfrett, 2021-01-04:
> Hi Dennis,
> thanks for the offer 👍 , but it is a rather seldom required functionality to have it switched dynamically, so we do not plan to implement it.

Thanks for the quick reply. Do you know from the top of your head in what function the actual pulsing & modulation code is executed?

ArminJo, 2021-01-04:
Best is to search for IRSND_GENERATE_NO_SEND_RF (in IRTimer.cpp.h)

closed 2021-01-04 by dennisfrett

dennisfrett, 2021-01-11:
For people looking for this, I've just whipped together a mini-library that does this: https://github.com/dennisfrett/Arduino-Direct-NEC-Transmitter