IRMP-org/IRMP #42: only picking up every 10-20 key press on Bang & Olufsen Beo4

CheapB, 2021-02-02:
## Bug Report

### Arduino Platform
* [ ] Arduino ATmega328* board (UNO, Nano)
* [ ] Arduino ATmega2560 board (Mega)
* [ ] Arduino ATmega32U4 board (Leonardo)
* [ ] Arduino megaAVR board (NanoEvery)
* [ ] Arduino SAM board (Due)
* [ ] Arduino SAMD board (Zero, MKR*)
* [ ] ATtiny85 board (ATTinyCore by Spence Conde)
* [ ] Digispark board
* [ ] ATmega8 board
* [ ] ESP8266 board
* [ X] ESP32 board
* [ ] STM32 board with "STM32 Boards (selected from submenu)" core
* [ ] STM32 board with "STM32F1 Boards (STM32duino.com)" core
* [ ] Apollo3 board
* [ ] Teensy board
* [ ] Other - please specify

### IDE
* [ X] Arduino IDE
* [ ] Arduino Pro IDE
* [ ] Sloeber IDE
* [ ] PlatformIO IDE
* [ ] Other - please specify

### IR-Protocol
* [ ] Unknown
* [ ] Sony
* [ ] NEC
* [ ] Samsung
* [ ] RC5, RC6
* [ ] Kaseikyo
* [ ] Denon
* [ ] JVC
* [ ] Lego
* [ ] Matsushita
* [ ] Mitsubishi
* [ ] Panasonic
* [ ] Sanyo
* [ ] Sharp
* [ ] Telefunken
* [ X] Other - please specify Bang & Olufsen

### Example to reproduce the issue
* [ ] AllProtocols
* [ ] Callback
* [ ] Interrupt
* [ ] OneProtocol
* [ ] ReceiveAndSend
* [ ] ReceiveAndSendDynamicPins
* [ ] RFProtocols
* [ ] SimpleReceiver
* [ ] SimpleSender
* [X ] Other
SimpleSender modified to use irmpSelectAllProtocols.h

### Pin(s) used for IR-receive, if not default
default 
### Version
* [ x] Yes I use the [latest repo version](https://github.com/ukw100/IRMP/archive/master.zip) and verified this!
* [ ] Other - please specify

### Current behavior

SimpleSender with line 56 modified to #include <irmpSelectAllProtocols.h>  and 57 commented out

only every 10-20 key press is being detected

### Expected behavior

I expected every key press to be detected

### Error output

"P=BANG OLU0xE  A=0x0 C=0x80"      every 10-20 time I press TV on the remote

### Additional context

ArminJo, 2021-02-02:
> SimpleSender with line 56 modified to #include <irmpSelectAllProtocols.h> and 57 commented out

What are you telling??? line 56 is IRMP_DATA irsnd_data;

closed 2021-02-02 by ArminJo

CheapB, 2021-02-03:
I just downloaded the lib and in my editor line 56 is #include <irmpSelectMain15Protocols.h>  // This enables 15 main protocols

not sure if that has changed over the last 4 hours...

ArminJo, 2021-02-03:
And what you think is this? https://github.com/ukw100/IRMP/blob/4e9ef0d7ed169965943485a84f7fb295e668c797/examples/SimpleSender/SimpleSender.ino#L56

CheapB, 2021-02-03:
Looks like I mistakenly marked and referenced SimpleSender instead of SimpleReceiver, Which my issue was referring to. You are obviously annoyed with my mistake - which i understand since you are offering your service for free to the community. I