IRMP-org/IRMP #43: Benchmark Media protocol

lpaolini, 2021-02-02:
I would like to add support for the [Benchmark remote control](https://benchmarkmedia.com/collections/parts/products/benchmark-remote-control-metal), which doesn't seem to be supported by any of the existing protocol, but I don't know exactly how to do that.

I've used a logic analyzer and measured the following timings:

- first bit pulse: 2.59ms
- first bit pause: 2.59ms
- pulse: 1.20ms
- short pause: 1.11ms
- long pause: 2.84ms
- data: 16bit
- stop bit: yes (pulse only)
- fsb/lsb: unknown
- interframe pause: 58ms

Do they resemble some known protocol?
If not, would it be possible for you to add support for it?
Perhaps you could create a branch for that, so that I can test it, and then submit a PR.

[image]
[image]
[image]
[image]

ArminJo, 2021-02-03:
Best is to use https://github.com/Arduino-IRremote/Arduino-IRremote use ir_Template.cpp or try to modify the ir_JVC.cpp timing. Then check if it is detected.
Best is if timing analysis can be done at the send diode, this removes the receiver pulse length distortions.

lpaolini, 2021-02-03:
Ok thanks, I'll give it a try.
Do you think you'll add support for Teensy 4.x?

ArminJo, 2021-02-03:
Sorry but there is no Teensy support possible, see: https://forum.pjrc.com/threads/65912-Enable-Continuous-Integration-with-arduino-cli-for-3-party-libraries 😞
But you can ask Paul, if he is willing to support 3. Party library developer better, than now.

closed 2021-02-10 by ArminJo