IRMP-org/IRMP #44: Bang and Olufsen send frequency seems to be wrong, and send is not working

chrwh, 2021-02-13:
## Bug Report

### Board
* [ ] Arduino ATmega328* board (UNO, Nano)
* [ ] Arduino ATmega2560 board (Mega)
* [ ] Arduino ATmega32U4 board (Leonardo)
* [ ] Arduino megaAVR board (NanoEvery)
* [ ] Arduino SAM board (Due)
* [ ] Arduino SAMD board (Zero, MKR*)
* [ ] ATtiny85 board (ATTinyCore by Spence Conde)
* [ ] Digispark board
* [ ] ATtiny3217 board (TinyCore)
* [ ] ATmega8 board
* [ ] ESP8266 board
* [X] ESP32 board (M5STACK)
* [ ] STM32 board with "STM32 Boards (selected from submenu)" core
* [ ] STM32 board with "STM32F1 Boards (STM32duino.com)" core
* [ ] Apollo3 board
* [ ] Teensy board
* [ ] Other - please specify

### IDE
* [X] Arduino IDE V.1.8.13
* [ ] Arduino Web Editor
* [ ] Arduino Pro IDE
* [ ] Sloeber IDE
* [ ] PlatformIO IDE
* [ ] Other - please specify

### IR-Protocol
* [ ] Unknown
* [ ] Sony
* [ ] NEC
* [ ] Samsung
* [ ] RC5, RC6
* [ ] Kaseikyo
* [ ] Denon
* [ ] JVC
* [ ] Lego
* [ ] Matsushita
* [ ] Mitsubishi
* [ ] Panasonic
* [ ] Sanyo
* [ ] Sharp
* [ ] Telefunken
* [X] Other - please specify (Bang and Olufsen)

### Example to reproduce the issue
* [ ] AllProtocols
* [ ] Callback
* [ ] Interrupt
* [ ] OneProtocol
* [X] ReceiveAndSend
* [ ] ReceiveAndSendDynamicPins
* [ ] RFProtocols
* [ ] SimpleReceiver
* [ ] SimpleSender
* [ ] Other

### Pin(s) used for IR-receive, if not default
IR Receive pin 19 (Using TSOP7000)
IR Send pin 16 (IR led 940nm I know that the IR LED is working, since I use some other code now for sending to the B&O, but would like to port to IRMP)

### Version
* [ ] Yes I use the [latest repo version](https://github.com/ukw100/IRMP/archive/master.zip) and verified this!
* [X] Other - please specify  (library version 3.4.0)

### Current behavior

[IRMP_ReceiveAndSend.zip](https://github.com/ukw100/IRMP/files/5974962/IRMP_ReceiveAndSend.zip)

The receive is working without any issues, but I can't get the send to work. If I look at the output with my Logic analyser the frequency is wrong it's about 117KHz

### Expected behavior

This is just a send test, 
If button A is clicked, send something
If button A is clicked, send something else

### Error output

No errors are outputed
### Additional context

ArminJo, 2021-02-13:
Maybe the call `getApbFrequency()` in the line 
https://github.com/ukw100/IRMP/blob/36a4c4106fa0ae9054a6770c8ef7f542ac1091b5/src/IRTimer.cpp.h#L238
returns the "wrong" value.
try (getApbFrequency() / (80 ***4**)) and please report your experience.

chrwh, 2021-02-13:
Hi,
That did not seems to help, the output freq is still about 117.6kHz at the max, but also alot of 21.51kHz stuff.
I might add, that if I try the same with Samsung protocol, the output freq is approx. 38KHz
Best regards,
Christian

chrwh, 2021-02-13:
Also the receiver stops working, when implementing the suggested above.

chrwh, 2021-02-24:
Any news regarding this?

ArminJo, 2021-02-24:
I am sorry but generating 455 kHz output is not supported by IRMP. I now documentetd it more clearly.

commit bf13675c (2021-02-24): Ducumernted that send IR frequency is fixed at 38 kHz. Closes #44
closed 2021-02-24 by ArminJo