IRMP-org/IRMP #48: Arduino simulations for the examples 😀 - feature update

Puneethkumarsn, 2021-05-18:
Hello there

huge thanks to the wonderful libraries. 
I have been playing with a simple receiver example on the wokwi Arduino simulator(https://github.com/wokwi) and glad that I found these examples.

My request: 
I would love to create a collection of examples(with all attributes without failing to mention) using your library on Wokwi Arduino Simulator for all our readers (https://wokwi.com)

**Example link:** https://wokwi.com/arduino/projects/298945438795432456

I would like to know If you are interested in the online simulation links (as above) for your readers as well. If yes, I will also share the table (examples and the links to the examples). All these links are permanent links. 
Thanks a lot for your time. 

[image]

ArminJo, 2021-05-18:
This is an awesome tool!!!
I would be very pleased to include such project links in the our project readme.
Thanks a lot!
Armin

Puneethkumarsn, 2021-05-19:
I am glad you liked it. The simulator is completely free for all.
I will soon create more examples and if I am stuck, will contact you. i hope you don't mind,

Thanks a lot again
Best Regards

Puneeth

Puneethkumarsn, 2021-05-19:
Dear Armin

Good day

I have created three examples quickly -  I would love to hear your feedback

Are these three examples works as expected?

**ReceiverTimingAnalysis** - https://wokwi.com/arduino/projects/299033930562011656
**minimal recevier** - https://wokwi.com/arduino/projects/299034264157028877
**simpleReceiver** - https://wokwi.com/arduino/projects/298945438795432456

What I realised is, we don't have an IR transmitter in the simulator (but we are already working on it). Hence, I will create more examples soon after we get the IR transmitter part up and running.

Once again, thanks for the opportunity. 
Cheers!
Puneeth

Discord link to the simulator - https://wokwi.com/discord

ArminJo, 2021-05-24:
Dear Puneeth

thanks again!!!
All of these work perfect!

I added the links to the examples in my 2 IR repositories.
https://github.com/ukw100/IRMP#examples
and
https://github.com/Arduino-IRremote/Arduino-IRremote#examples

Best regards
Armin

closed 2021-05-24 by ArminJo