IRMP-org/IRMP #51: CANNOT decode RC6

YUDHPK, 2021-06-10:
If you have handling problems or questions, consider to open a discussion https://github.com/Arduino-IRremote/Arduino-IRremote/discussions instead of an issue.

## Bug Report

### Board
* [x] Arduino ATmega328* board (UNO, Nano)

### IDE
* [x] Arduino IDE

### IR-Protoco
* [x] RC5, RC6
* [x] Panasonic

### Example to reproduce the issue
* [x] AllProtocols
* [x] OneProtocol

### Version
* [x] Yes I use the [latest repo version](https://github.com/Arduino-IRremote/Arduino-IRremote/archive/master.zip) and verified this!

Please delete all unchecked lines above :-)

### Pin(s) used for IR-receive, if not default

### Current behavior

i am unable to decode tatasky remote's IR signal which uses RC6 Protocol using your code.
i found the same codes using IRremote library

Also i cannot decode Panasonic AC's IR code as well as of OGeneral(Fujitsu) AC's 
### Expected behavior

Also while enabling Panasonic codec initially i had trouble because it didnt display a compile time warning in SelectALLProtocols example to disable KASEIKYO protocol .
### Additional context

**checklist:**
- [x] I have **read** the README.md file thoroughly.
- [] I have searched existing issues to see if there is anything I have missed.
- [x] I have browsed the examples for one, that matches my use case.
- [] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here.
- [] The title of the issue is helpful and relevant.
- [] I checked, if at least one of the examples was working.

** We will start to close issues that do not follow these guidelines as it doesn't help the contributors who spend time trying to solve issues if the community ignores guidelines!**

ArminJo, 2021-06-10:
> i found the same codes using IRremote library

So you are lucky!

> Also i cannot decode Panasonic AC's IR code as well as of OGeneral(Fujitsu) AC's

Try IRremote for this

> **it** didnt display

???

YUDHPK, 2021-06-11:
I initially used IRremote but consumed a lot of memory so when i came to know about IRMP Library i tried to use this library but encountered issue with an RC6 Tatasky remote.
What my project needs is only decoding constantly in hex format and displaying it on lcd like the SelectAllProtocols example. But i must be able to decode rc6 along with it. And ac remotes if possible.
So please help what should i do?
If you need rc6 value list i can provide it just tell me in which format? I will try my best

YUDHPK, 2021-06-11:
Also when i modified the select all protocols example to show the no. Of protocols activated it showed 44 and 47 with F_interrupts 20000 and Merlin

ArminJo, 2021-06-11:
You can post the IRremote long timing output of the protocols in question.

YUDHPK, 2021-06-17:
I will try as soon as possible but currently my laptop's hard disk crashed

closed 2022-04-06 by ArminJo