IRMP-org/IRMP #54: Additional documentation for DigiSpark boards

drechc, 2021-06-20:
## Bug Report

### Board
* [x] Digispark board

### IDE
* [x] Arduino IDE

### IR-Protocol
* [x] NEC

### Example to reproduce the issue
* [x] SimpleReceiver

### Version
* [x] Yes I use the [latest repo version](https://github.com/Arduino-IRremote/Arduino-IRremote/archive/master.zip) and verified this!

### Current behavior
None of the examples successfully compile using the "official" (but outdated) Digistump core.  (Various errors due to different compiler versions, missing pin definitions, etc).

When using the ATtinyCore by SpenceKonde with the "Micronucleus/Digispark Pro" board and the "DigiSpark Pro" pin mapping, the functions in digitalWriteFast.h do not use the same pin numbers as the rest of the core.

### Expected behavior
Documentation should include a reference to the newer and better DigiSpark core at:
https://github.com/ArminJo/DigistumpArduino

It would also be helpful to note the "New Style" pin mapping must be selected when using the ATtinyCore and the "Micronucleus/Digispark Pro" board.

**checklist:**
- [x] I have **read** the README.md file thoroughly.
- [x] I have searched existing issues to see if there is anything I have missed.
- [x] I have browsed the examples for one, that matches my use case.
- [x] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here.
- [x] The title of the issue is helpful and relevant.
- [x] I checked, if at least one of the examples was working.

** We will start to close issues that do not follow these guidelines as it doesn't help the contributors who spend time trying to solve issues if the community ignores guidelines!**

ArminJo, 2021-06-20:
Thanks for the reporting! Fixed by modifying digitalWriteFast.h

closed 2021-06-20 by ArminJo

drechc, 2021-06-21:
That's a neat trick, I couldn't think of a clean way to to do it...  Thanks for the fix!

commit 9b553b23 (2021-06-20): Modifying digitalWriteFast.h to be compatible with ATTinyCore Digispark Pro default pin mapping. Closes #54