IRMP-org/IRMP #55: Arduino Due compatibility

gleanlux, 2021-06-24:
Hello, 

I would like to inquire, is it possible to use this library on an Arduino DUE board?
The Arduino IDE compiler say, it's not compatible with "sam" boards.

Thank you very much for their help!

ArminJo, 2021-06-24:
SAM is not listed here https://github.com/ukw100/IRMP#supported-arduino-architectures--cpus--boards

closed 2021-06-24 by ArminJo

gleanlux, 2021-06-24:
Yes, I understand, but expected to be included in the grant?

ArminJo, 2021-06-24:
> but expected to be included in the grant?

I do not understand this sentence!

gleanlux, 2021-06-24:
Sorry, my english isn't perfect.

I would like to same as, in other library.
I would like the library to support DUE.

ArminJo, 2021-06-24:
DUE is a very poor supported platform 😞 .
But you are invited to try it.