IRMP-org/IRMP #63: esp8266 resets in non intrrupt mode

clear-sky, 2021-09-12:
If you have handling problems or questions, consider to open a discussion https://github.com/Arduino-IRremote/Arduino-IRremote/discussions instead of an issue.

## Bug Report

### Board
* [ ] Arduino ATmega328* board (UNO, Nano)
* [ ] Arduino ATmega2560 board (Mega)
* [ ] Arduino ATmega32U4 board (Leonardo)
* [ ] Arduino megaAVR board (NanoEvery)
* [ ] Arduino SAM board (Due)
* [ ] Arduino SAMD board (Zero, MKR*)
* [ ] ATtiny85 board (ATTinyCore by Spence Conde)
* [ ] Digispark board
* [ ] ATtiny3217 board (TinyCore)
* [ ] ATmega8 board
* [ y] ESP8266 board
* [ ] ESP32 board
* [ ] STM32 board with "STM32 Boards (selected from submenu)" core
* [ ] STM32 board with "STM32F1 Boards (STM32duino.com)" core
* [ ] Apollo3 board
* [ ] Teensy board
* [ ] Other - please specify

### IDE
* [y ] Arduino IDE
* [ ] Arduino Web Editor
* [ ] Arduino Pro IDE
* [ ] Sloeber IDE
* [ ] PlatformIO IDE
* [ ] Other - please specify

### IR-Protocol
* [ ] Unknown
* [ ] Sony
* [y ] NEC
* [ ] Samsung
* [ ] RC5, RC6
* [ ] Kaseikyo
* [ ] Denon
* [ ] JVC
* [ ] Lego
* [ ] Matsushita
* [ ] Mitsubishi
* [ ] Panasonic
* [ ] Sanyo
* [ ] Sharp
* [ ] Telefunken
* [ ] Other - please specify

### Example to reproduce the issue
* [ ] AllProtocols
* [ ] Callback
* [ ] Interrupt
* [y ] OneProtocol
* [ ] ReceiveAndSend
* [ ] ReceiveAndSendDynamicPins
* [ ] RFProtocols
* [ ] SimpleReceiver
* [ ] SimpleSender
* [ ] Other

### Version
* [ y] Yes I use the [latest repo version](https://github.com/Arduino-IRremote/Arduino-IRremote/archive/master.zip) and verified this!
* [ ] Other - please specify

Please delete all unchecked lines above :-)

### Pin(s) used for IR-receive, if not default
gpio 12
### Current behavior
in itrrupt mode esp8266 work well but when i try it in non intrrupt mode mcu resets and i get this message in serial console

START OneProtocol.ino from Sep 12 2021
Using library version 3.4.1

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v00042bd0
~ld

### Expected behavior
as far as i know main problem is in this function:
irmp_init();

### Additional context

i am trying to make a dotmatrix clock based on this project
https://github.com/tehniq3/ESP8266-LED-Matrix-Clock
but i have removed dht22 and replaced a vs838 ir sensor
ds3231 and ldr remains 
and this is vs838 wiring 
http://j5d2v7d7.stackpathcdn.com/wp-content/uploads/2016/11/VS838-application-circuit.gif

**checklist:**
- [y] I have **read** the README.md file thoroughly.
- [y] I have searched existing issues to see if there is anything I have missed.
- [y] I have browsed the examples for one, that matches my use case.
- [] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here.
- [] The title of the issue is helpful and relevant.
- [] I checked, if at least one of the examples was working.

** We will start to close issues that do not follow these guidelines as it doesn't help the contributors who spend time trying to solve issues if the community ignores guidelines!**

clear-sky, 2021-09-15:
wow!!!
previous  version works
this is serial output:
START OneProtocol.ino from Sep 15 2021
Using library version 3.4.0
Ready to receive IR signals of protocols: NEC, APPLE, ONKYO, at pin 12
P=NEC0x2 A=0xFD01 C=0x83
P=NEC0x2 A=0xFD01 C=0x83 R
P=NEC0x2 A=0xFD01 C=0x93

ArminJo, 2021-09-16:
Thanks a lot for hunting the bug.
It was the change from
`bool ICACHE_RAM_ATTR irmp_ISR(void)`
to 
`ICACHE_RAM_ATTR  bool irmp_ISR(void)`
that introduced the error

closed 2021-09-16 by ArminJo

clear-sky, 2021-09-17:
thanks for notice but problem has not been solved yet 
please see serial output:

START OneProtocol.ino from Sep 17 2021
Using library version 3.5.0

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v00042bd0
~ld

clear-sky, 2021-09-19:
in version 3.5.1 problem has been solved!
very very thanks for your notice!

ArminJo, 2021-09-19:
Thanks for reporting !

commit 764b443c (2021-09-16): void ICACHE_RAM_ATTR instead of ICACHE_RAM_ATTR void.  Closes #63
commit b6cfe541 (2021-09-17): void ICACHE_RAM_ATTR instead of ICACHE_RAM_ATTR void.  Closes #63