IRMP-org/IRMP #70: Defining NO_LED_FEEDBACK_CODE breaks compiliation

eku, 2022-05-28:
## Bug Report

### Board
* [x] ESP8266 board

### IDE
* [x] PlatformIO IDE

### Example to reproduce the issue
* [x] Other

### Version
* [x] Yes I use the [latest repo version](https://github.com/Arduino-IRremote/Arduino-IRremote/archive/master.zip) and verified this!

### Current behavior

```
#define IRMP_IRSND_ALLOW_DYNAMIC_PINS 1
#define USE_ONE_TIMER_FOR_IRMP_AND_IRSND 1
#define NO_LED_FEEDBACK_CODE
#include <irmp.hpp>
```
Compile error

```
lib/lib_div/IRMP/src/irmpArduinoExt.hpp: In function 'void irmp_init(uint_fast8_t, uint_fast8_t, bool)':
lib/lib_div/IRMP/src/irmpArduinoExt.hpp:35:5: error: 'irmp_irsnd_LedFeedbackPin' was not declared in this scope
     irmp_irsnd_LedFeedbackPin = aFeedbackLedPin;
     ^
lib/lib_div/IRMP/src/irmpArduinoExt.hpp:36:5: error: 'irmp_irsnd_LedFeedbackPinIsActiveLow' was not declared in this scope
     irmp_irsnd_LedFeedbackPinIsActiveLow = aIrmpLedFeedbackPinIsActiveLow;
     ^
lib/lib_div/IRMP/src/irmpArduinoExt.hpp: In function 'void irmp_init(uint_fast8_t, uint_fast8_t)':
lib/lib_div/IRMP/src/irmpArduinoExt.hpp:59:47: error: 'irmp_irsnd_LedFeedbackPinIsActiveLow' was not declared in this scope
     irmp_init(aIrmpInputPin, aFeedbackLedPin, irmp_irsnd_LedFeedbackPinIsActiveLow);
```

**checklist:**
- [x] I have **read** the README.md file thoroughly.
- [x] I have searched existing issues to see if there is anything I have missed.
- [] I have browsed the examples for one, that matches my use case.
- [x] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here.
- [x] The title of the issue is helpful and relevant.
- [] I checked, if at least one of the examples was working.

** We will start to close issues that do not follow these guidelines as it doesn't help the contributors who spend time trying to solve issues if the community ignores guidelines!**

closed 2022-05-28 by ArminJo

eku, 2022-05-28:
Thanks for quick responese.

If `NO_LED_FEEDBACK_CODE` is defined, `irmp_irsnd_LEDFeedback` should not be available, as used in `OneProtocol.ino`.

ArminJo, 2022-05-28:
Thanks for reporting !!! 🥇

eku, 2022-05-28:
Still compile errors, even with 62c8689faf036899462709cc1a017f3fb9943539.

ArminJo, 2022-05-28:
You mean this **Must not be enabled** ?
<html><body>

USE_ONE_TIMER_FOR_IRMP_AND_IRSND | disabled | Must be defined if you use receiving and sending simultaneously, since both use the same timer resource. Must not be enabled if you only use receiving.
-- | -- | --

</body>
</html>

eku, 2022-05-28:
I use the receiver and sender simultaneously, so I defined `USE_ONE_TIMER_FOR_IRMP_AND_IRSND`. I hope I got that one right.

Compile errors still come from defining `NO_LED_FEEDBACK_CODE`.
```
lib/lib_div/IRMP/src/irsndArduinoExt.hpp: In function 'void irsnd_on()':
lib/lib_div/IRMP/src/irsndArduinoExt.hpp:121:13: error: 'irmp_irsnd_LedFeedbackEnabled' was not declared in this scope
         if (irmp_irsnd_LedFeedbackEnabled)
             ^
lib/lib_div/IRMP/src/irsndArduinoExt.hpp:124:43: error: 'irmp_irsnd_SetFeedbackLED' was not declared in this scope
             irmp_irsnd_SetFeedbackLED(true);
                                           ^
lib/lib_div/IRMP/src/irsndArduinoExt.hpp: In function 'void irsnd_off()':
lib/lib_div/IRMP/src/irsndArduinoExt.hpp:150:13: error: 'irmp_irsnd_LedFeedbackEnabled' was not declared in this scope
         if (irmp_irsnd_LedFeedbackEnabled)
             ^
lib/lib_div/IRMP/src/irsndArduinoExt.hpp:153:44: error: 'irmp_irsnd_SetFeedbackLED' was not declared in this scope
             irmp_irsnd_SetFeedbackLED(false);

```

ArminJo, 2022-05-28:
I amended the last commit,  so you must get the sources again.

eku, 2022-05-28:
`git pull` says `Bereits aktuell.`.

ArminJo, 2022-05-29:
Sorry sorry! I checked the wrong example.

ArminJo, 2022-05-29:
Hi Erik,
can you test it again, before I release the hotfix.
Thanks!

eku, 2022-05-29:
Both IRMP and IRSND compile with `NO_LED_FEEDBACK_CODE`. 

I think I have seen potential for optimisation. Possibly make the including of headers dependent on this. But it's up to you.

ArminJo, 2022-05-29:
Thanks 👍 
Code size optimisations are always welcome 😀

commit 9332e7b1 (2022-05-28): Housekeeping and closes #70
commit 62c8689f (2022-05-28): Housekeeping and closes #70
commit d42c006b (2022-05-29): Housekeeping and closes #70