IRMP-org/IRMP #94: Compile Fail with ESP32

tweisberger, 2024-05-09:
If you have handling problems or questions, consider to open a discussion https://github.com/IRMP-org/IRMP/discussions instead of an issue.

## Bug Report

### Board
* [x] ESP32 board -- Specifically S3

### IDE
* [x] Arduino IDE

### IR-Protocol
* [x] NEC

### Example to reproduce the issue
* [x] AllProtocols
* [x] Callback
* [x] ReceiveAndSend
* [x] SimpleReceiver

### Version
* [x] Yes I use the [latest repo version](https://github.com/Arduino-IRremote/Arduino-IRremote/archive/master.zip) and verified this!

Please delete all unchecked lines above :-)

### Pin(s) used for IR-receive, if not default

### Current behavior

Using any of the demos as is.

[IRMP_compile_error.txt](https://github.com/IRMP-org/IRMP/files/15265646/IRMP_compile_error.txt)

Compilation failure

### Expected behavior

Compiles and runs

### Additional context

ArminJo, 2024-05-10:
I cannot reproduce it

closed 2024-05-10 by ArminJo

tweisberger, 2024-05-10:
@ArminJo What version of the ESP32 core are you using?

For me: 3.0.0-rc1, in Arduino IDE 2.3.2
```
Using board 'lolin_s3_mini' from platform in folder: /Users/user/Library/Arduino15/packages/esp32/hardware/esp32/3.0.0-rc1
Using core 'esp32' from platform in folder: /Users/user/Library/Arduino15/packages/esp32/hardware/esp32/3.0.0-rc1
```
 It works with ESP 8266, but not with ESP32. In IRTimer.hpp, line 265, there is a function call `timerBegin(3, 80, true);` This is an error, since function in the Arduino library: packages/esp32/hardware/esp32/3.0.0-rc1/cores/esp32/esp32-hal-timer.h:
```
hw_timer_t * timerBegin(uint32_t frequency);
```

tweisberger, 2024-05-10:
This appears to be an issue when using the Espressif ESP32 3.0.0 core, I downgraded to 2.0.16, and it now works. I would argue that this is an issue for users using 3.0.0, which the Arduino IDE automatically selects, even though it is still in RC phase.